pycom/pymakr-vsc #1: No REPL with COM port connection

Thosch42, 2017-08-01:
Hi all,
Pymakr plugin is simply not working. The Pycom Console button on the bottom line toggles by clicking on it between cross and hook, regardless if there is a device connected to the computer or not. The "List Serial Ports" command, does just nothing.
My box works on Windows 10 x64. My Pymakr plugin for Atom is also broken. Maybe I go back to something working, till the Pymakr tools work properly.

Cheers,
Thomas

RalphHogenbirk, 2017-08-01:
Hi @Thosch42, sorry to hear that. I've had two similar cases of this in vscode, where the terminal just remained black. Problem there was that node was not installed, which is needed for the terminal (which runs in a process outside of VSCode). Try installing node and adding it to your path.

What's broken about Atom exactly? Maybe I can help you there as well (or add a new ticket on the atom repo if it's an extended problem)

Thosch42, 2017-08-01:
Hi Ralph,
ok, very good information, that I need to have node installed. Will do this and come back.

With the Atom plugin I have the same problem like already described in https://github.com/pycom/pymakr-atom/issues/24
I uninstalled, reinstalled Atom, the plugin, everything force and back, but get always just the red box.

Cheers,
Thomas

RalphHogenbirk, 2017-08-01:
Ok let me know.

About the atom plugin, check if you get python version 2.7 when doing `apm version`. If not, run: 
`echo "python=/usr/bin/python2.7" >> ~/.atom/.apmrc` to tell apm to use python 2.7. A [similar issue](https://github.com/pycom/pymakr-atom/issues/25) was solved this way.

Thosch42, 2017-08-01:
Hi Ralph, thanks a lot. Now it runs like a charm.

Cheers,
Thomas

closed 2017-08-01 by Thosch42