pycom/pymakr-vsc #2: Plugin terminal problem

leogaggl, 2017-08-21:
Just moving this issue from the pymakr-atom repo as I created in the wrong repo.

History: [https://github.com/pycom/pymakr-atom/issues/31](https://github.com/pycom/pymakr-atom/issues/31)

Still having the problem that I can only connect via WIFI. Has anybody been able to connect to the Pycom device (in my case a LoPy) via USB Serial ?

gmires, 2017-08-21:
hi @leogaggl ,
This way i connect you to the lopy via ssh eth.
Have you changed the configuration of pymakr plugin to make sure you do not connect to an ip address but to the serial device ??

(ex. /dev/ttyUSB0) in pymakr general option or project option.
In the box where at this time there should be the lopy ip address

see this
https://forum.pycom.io/assets/uploads/files/1502785891182-screen-shot-2017-08-15-at-4.30.28-pm.png

regards

leogaggl, 2017-08-22:
Thanks @gmires 

"address": "/dev/ttyUSB0" in the settings file pymakr.json did the trick for me. 

Thanks for your help !

closed 2017-08-22 by leogaggl