pycom/pymakr-vsc #3: Pymakr does not work on VSCode x64 on Windows

psotres, 2017-08-24:
Console shows:

Activating extension `pycom.Pymakr` failed: %1 is not a valid Win32 application.
\\?\C:\Users\myuser\.vscode\extensions\pycom.pymakr-0.1.1\precompiles\serialport-win32\build\Release\serialport.node

However it is working on VSCode x32.

ditochino, 2017-08-31:
I fixed this issue by downloading the serialport package via npm and pasting it into the pymakr extension:

https://www.npmjs.com/package/serialport

RalphHogenbirk, 2017-09-06:
Are you using the x64 version of VSCode as well, or the 32bit version? Turns out the plugin doesn't handle the 64 install of vscode. We'll publish a new version this week with support for it.

RalphHogenbirk, 2017-09-12:
This has been fixed on friday, and today there was an update to support 1.16 as well on the other OS's. Let me know if this issue can be closed.

closed 2017-12-22 by RalphHogenbirk