pycom/pymakr-vsc #6: Extension gets in the way when working on other projects

brocaar, 2017-12-01:
Hi, first of all thanks for providing this extension. After using it a bit I would like to share some feedback. I'm using VSCode for different kinds of projects (Python, Go, JS, HTML, ...).

When working on non-Pycom (LoPy) related projects, I experience that this extension takes over VSCode a bit too much and gets in my way. My current flow is that I disable this extension completely, and only enable it manually when working on Pycom related code. This should not be needed.

What I would like is:

* Remove the extension buttons and menu's from the status bar
* That this extension does not change the default behavior of opening the terminal window (when working on a `*.go` file, I don't want the Pycom terminal, I want a normal shell)

I think it would be sufficient when the extension shows the available commands in the Command Palette when the editor is in the context of a `*.py` file and that people are able to open a Pycom terminal through this Command Palette, instead of changing this default behavior.

RalphHogenbirk, 2017-12-07:
Hi @brocaar, thanks for your feedback! it's appreciated. I didn't hear any feedback before this, that people thought the buttons on the bottom were in the way. But I can imagine why and maybe there are more people with this problem that just didn't bother to report it yet.

First about the buttons from the menu bar: do you have other buttons there that conflict with the pymakr buttons? In what way are you bothered by the presence of the 4 pymakr buttons? I think they are very useful when you are working with a pycom board and project. You tend to use them a lot when working and testing (especially the sync and run) and they are much much more convenient than using the command palette every time. Plus, I have no other place to display the current connection status of the board. 
One way to solve this would be to add an optional global config variable 'display_statusbar_buttons' so you can disable them when they bother you and you just want the command palette options. Would that suite you?

Then the terminal behaviour: it only changes the behaviour the first time. It creates the Pycom console and switches to it automatically. After this, you can switch back to the normal shell with the dropdown on the right. Then it will stay on the normal shell whatever you do, except for when you use a Pymakr feature that has to output to the pymakr terminal. Or does it bother you in another way? Do you restart VSC so often that you have to switch back to the normal shell a lot?

Let me know what you think! Lets try to find a solution that works just as convenient as it does now, but doesn't get in your way when working with other projects.

brocaar, 2017-12-07:
> First about the buttons from the menu bar: do you have other buttons there that conflict with the pymakr buttons? In what way are you bothered by the presence of the 4 pymakr buttons?

I don't have any other buttons and actually that is the way I like it ;-) Most of the time I work on non Pycom related projects so for me personally there is no need why these buttons should be present (all the time). Most of the times when I perform an action (e.g. running tests) I use the command pallet and I prefer that workflow over extensions adding buttons to the VSCode UI. However a `display_statusbar_buttons` could be an option I think to make it work for different kind of workflows.

> Then the terminal behaviour: it only changes the behaviour the first time. It creates the Pycom console and switches to it automatically. After this, you can switch back to the normal shell with the dropdown on the right.

But that is directly my issue, I need to switch back in order to get a terminal. I would prefer to use `shift + command + P` to open the Pycom terminal when I need it. In any other case, I expect to just get the terminal.

I work on quite a few different projects and usually I open a VSCode window by executing `code .` from an external terminal. Each time when this opens a new VSCode window, I need to close the Pycom terminal an switch back to a normal terminal. Because of this behavior, I now have the extension disabled by default and only enable it when I need it (which should not be the final solution I think).

RalphHogenbirk, 2017-12-08:
Ok thanks for the feedback and info. I'll keep this all in mind when working on the next update of the plugin. I'd probably implement the config option for the statusbar. Possibly even as an array so you can decide which buttons you want and which you don't. 

I'll try to look at the terminals default behaviour to get it out of your way when opening the terminal for the first time. Probably I'll create it, but keep the standard shell selected until you use one of the pymakr features. I'll have to test if I can control that well enough though to not have people searching for it when they do need it.

RalphHogenbirk, 2017-12-29:
@brocaar In the latest version (0.1.7) 've added some options to the config that might be useful for you:

You can add a 'statusbar_buttons' key to your global config file to define which buttons you get to see in the status bar. Make it an empty array to remove all buttons (only 'all commands' will stay active). All available options are in the readme page of this repo.

The terminal behaviour could already be changed in the config in ealier versions. Just set 'open_on_start' to false and it won't open anymore until you use any of the pymakr features.

Let me know if that works for you

closed 2019-05-16 by RalphHogenbirk