pycom/pymakr-vsc #7: Unable to synchronise project

pjb304, 2018-01-17:
Hi,

I'm trying to update a project I had previously running on a LoPy board and am currently unable to sync the project from Visual Studio Code.

I have updated my version of node.js to the current LTS (8.9.4), and am running Visual Studio Code 1.19.2, Date 2018-01-10 Shell 1.7.9 Renderer 58.0.3029.110, Node 7.9 Architecture x64 with Pymakr 0.1.7.  I have also run the firmware updater on the LoPy device and it is running the latest firmware.  I am running Visual Studio code under 64bit Windows 10.

Before performing all the updates today I could not connect to a LoPy at all.  Getting the error "command 'pymakr.connect' not found". Having performed the updates to the above version I can now connect to a LoPy device sucessfully, however, I am currently unable to synchronise the project and get the error "command 'pymakr.sync' not found.

Do you have any suggested solutions to this problem?  Many thanks.

RalphHogenbirk, 2018-01-17:
Hi @pjb304, good to hear the old issue is solved. I double checked the code and there was a mistake in the last update. The 'sync' command was changed to 'upload' in the last updates, but the command that is used in the shortkey and context menu was not changed properly. I'll release a version soon with that fixed.

For now, use only the 'upload' button on the bottom left to upload your code to the board.

pjb304, 2018-01-18:
Hi @RalphHogenbirk, many thanks for the prompt response. Using upload instead of sync does indeed behave as expected.

yjvben, 2018-05-13:
Original problem still exists
the command that is used in the shortkey and context menu was NOT changed.
using visual code 1.23.1

RalphHogenbirk, 2018-05-14:
Hi @yjvben, there still hasn't been a release for VSC. It's lacking behind on the Atom version a little at the moment (though it's working fine except for the shortkey). We have a big new release almost ready now for VSC. Sorry for the delays on this.

closed 2019-05-16 by RalphHogenbirk