pycom/pymakr-vsc #8: Upload Failed

TamiasSibiricus, 2018-02-03:
OS: Ubuntu 16.04
Python 2.7
PyMakr VScode extension 0.1.7
Reason: there is no possible upload file to board.

```
Connecting on /dev/ttyUSB0...

>>> import esp
>>> esp.check_fw()
size: 600872
md5: 35e285a80516e70242ebf7d780d6c70f
True
>>>
Uploading project (main folder)...

Reading file status
Failed to read project status, uploading all files
[1/2] Writing file boot.py
Upload failed. Please reboot your device manually.

MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>>
```

Seems this bug is very similar to https://github.com/pycom/pymakr-atom/issues/58

RalphHogenbirk, 2018-02-05:
Hi @TamiasSibiricus, it might indeed be similar.  VSCode hasn't been updated with the last bugfixes that Atom has. After the next atom release this week, we'll do an update of VSC as well, then you should try again. In any case, update the firmware to the very latest release, which might make some difference already.

TamiasSibiricus, 2018-02-05:
Thanks for response

> update the firmware to the very latest release
Stable or nightly prefered?

RalphHogenbirk, 2018-02-05:
Get the stable version, using the Pycom Firmware Update tool

closed 2018-04-23 by TamiasSibiricus