pycom/pymakr-vsc #9: Unavailable to upload code to the device

sruiz36, 2018-06-18:
Hi :),
I'm using the Pymakr Plugin on Visual Code, I saw some changes were applied on pymakr.
However, I'm unavailable to upload the code to the ESP32 device, the terminal shows the following trouble.
[image]
I made some tries, but none of those were successful.
**Note:** using Atom editor, works fine!

Thanks in advance!

RalphHogenbirk, 2018-06-19:
Thanks for reporting. Pymakr for Atom is a bit ahead in development right now and VSC is still running on older upload logic (that has some flaws in it like this one) We're still improving some last issues in Atom, but once they are solved we'll update the VSCode plugin with the same code. That'll solve your issues. I'll let you know when we get that far.

One thing you can try is to do a safe-boot (ctrl-f) before pressing upload. If that doesn't help, I hope you're OK with using Atom for now.

RalphHogenbirk, 2019-05-16:
We've forgotten to ever close this issue, but I'm confident that this has been solved in the latest releases for vscode :)

closed 2019-05-16 by RalphHogenbirk