pycom/pymakr-vsc #13: command pymakr.connect not found

elektrobil, 2018-09-11:
I have still same error `:command pymakr.connect not found`
Im using VSCODE on Opensuse Tumbleweed   
Can you help for this issue ?

sdaniel55, 2018-09-11:
Hi @elektrobil 👋 

Could you tell me your VSCode and Pymakr version numbers? Have you tried reinstalling the pymakr plugin?

elektrobil, 2018-09-16:
2 days ago I downgraded 1.25.1 for using Pymakr and waiting for fixing this bug

johncaipa, 2018-09-21:
I have the same Issue on Ubuntu 18, extension don't work
VSCode version 1.27.2
Pymakr version 1.0.2

narrowtux, 2019-05-08:
I'm getting this error message again. 

VS Code Version 1.28.2 (1.28.2)
Pymakr 1.0.7

narrowtux, 2019-05-08:
nvm, my VS Code was outdated and apparently didn't auto update itself to 1.33

HeavensBlade, 2019-05-12:
I'm getting this error

VS Code 1.33.1
Pymakr 1.1.0

RalphHogenbirk, 2019-05-13:
@HeavensBlade Try to update to 1.1.2, see if that helps.

WaldemarCichon, 2020-09-14:
After update of VSC to Version: 1.49.0 and having PyMakr at version 1.1.6 I got this error. Prevoiusly the plugin was working flawlessly. 

The problem appears as well on Windows as also on Mac.

nielsclausen, 2020-09-14:
> After update of VSC to Version: 1.49.0 and having PyMakr at version 1.1.6 I got this error. Prevoiusly the plugin was working flawlessly.

I experience the same on v 1.49.0 (mac). [Downgraded](https://stackoverflow.com/questions/49346733/how-to-downgrade-vscode) to [VSCode 1.48.2](https://code.visualstudio.com/updates/v1_48) and all worked again.