pycom/pymakr-vsc #15: bug in disconnect() makes it impossible to re-connect without reloading vscode

Josverl, 2018-10-05:
## What are the steps to reproduce this issue?
1. start up vscode , connect a generic esp32 board 
2.  works as expected
3. physically unplug board , and re-attach

## What happens?
4. pycomm is not able to connect / disconnect any more 

## What were you expecting to happen?
I expect to be able to disconnect and reconnect 

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
VSCode shows an error and is unable to connect any more 
`X  Running the contributed command:'pymakr.toggleConnect' failed.` 

a rough workaround is to reload the VSCode window using `>Reload Window`

## Any other comments?
there appears to be a bug in the connect function; it references _this. while is should reference this.

`  
disconnect(cb){
    this.logger.info("Disconnecting...")
    if(this.pyboard.isConnecting()){
        this.terminal.writeln("Connection attempt canceled")
    }
    // else{
    //   this.terminal.writeln("Disconnected. Click here to reconnect.")
    // }
    clearInterval(this.connection_timer)
    this.api.setConnectionState(_this.pyboard.address,false) // <-- _this does not exists ….
    this.pyboard.disconnect(function(){
      if(cb) cb()
    })
    this.synchronizing = false
    this.runner.stop()
    this.setButtonState()
  }

`
## What versions of software are you using?

**Operating system:**
Windows 10 [Version 10.0.17134.285]

**VSCode version:**
Version: 1.27.2 (system setup)
Commit: f46c4c469d6e6d8c46f268d1553c5dc4b475840f
Date: 2018-09-12T16:17:45.060Z
Electron: 2.0.7
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
Architecture: x64

**Pymakr version:**
1.0.3

RalphHogenbirk, 2018-10-06:
Thanks for the report, that's quite an obvious bug. It fails on that line so it never tells this.pyboard to actually disconnect. Fixed it ont he develop branch, will include it in the next release.

Josverl, 2018-12-27:
Thanks  Ralph, confirmed fixed in 1.0.4

closed 2018-12-27 by Josverl