pycom/pymakr-vsc #16: board waiting in bootloader after reset

Josverl, 2018-10-05:
## What are the steps to reproduce this issue?
1. start vscode , connect generic ESP32 (M5Ssack. Lolin32Pro, others) 
2. connect to the board 
3. reset the board using a physical reset button 

## What happens?
* the board will reboot, but wait in the bootloader for a firmware download that never starts 
* This is likely caused by the board booting while USB / Serial DTR is High , in combination with an autodownload circuit 

## What were you expecting to happen?
* the board to boot normally 

## Any logs, error output, etc?

* The terminal shows something like : 
'
>>> ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x27 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
'

## Any other comments?
I have created a fix/workaround  for this by setting rts to high in pyserial - sendPing() for windows 
but I have not been able to test it on other platforms or with other devices 

## What versions of software are you using?

**Operating system:**
Microsoft Windows [Version 10.0.17134.285]

**VSCode version:**
Version: 1.27.2 (system setup)
Commit: f46c4c469d6e6d8c46f268d1553c5dc4b475840f
Date: 2018-09-12T16:17:45.060Z
Electron: 2.0.7
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
Architecture: x64

**Pymakr version:**
1.0.3

RalphHogenbirk, 2018-10-06:
If this fixes the behaviour for the generic ESP32 boards without affecting the Pycom board behaviour, I'm all fine with this. Will try to test this code next week.

Josverl, 2018-12-27:
confirmed fixed in 1.0.4 .

closed 2018-12-27 by Josverl