pycom/pymakr-vsc #17: Add : Run a line of code

Josverl, 2018-10-05:
# Feature Request 🚀

Allow a lines of code , or a block of code to be set to the board for execution,
similar to the Ctrl-Enter shortcut in VSCode + Python 

## Describe the solution you'd like
-Assign keybinding ( Ctrl-Enter / Ctrl-Shift-L or something) 
if the cursor is on a line 
> send that line to the board 
> execute the code 
> put focus back in the editor 

if a block is selected 
> send that block to the board ( must use paste-mode) 
> execute the code 
> put focus back in the editor 

this would allow a developer to test single lines, or block of code ,
and effectively single-step though a piece of code by running it line by line 

as a further enhancement, extraneous indentation could be removed to allow running if sub-sections of indented code 

## Describe alternatives you've considered
Copy and Paste from the editor to the terminal is possible , but error prone 
this is not an effective workflow

## Teachability, Documentation, Adoption, Migration Strategy
*If you can, explain how users will be able to use this and possibly write out a version the docs. Maybe a screenshot or design?*

I have sample working code in https://github.com/pycom/pymakr-vsc/compare/master...Josverl:RunByLine

RalphHogenbirk, 2018-10-06:
Hi @Josverl, thanks a lot for this feature request. Definitely a very good addition. Your code is also looking good!

Brainstorming on about this: The indentation should definitely be handled before adding the feature, since a selection of code will be indented 9 out of 10 times. Should not be too hard to do: 1) remove indents from 1st line of selection. 2) remove the same amount of indents from all the other lines. 
Other thing I'm thinking about is to add this feature to the normal 'run' command instead of adding a separate command. If the user has selected 1 line or more, it runs that selection. If not, it runs the whole file like it does now. 

I'll check it out after the weekend.

Josverl, 2018-10-06:
Hi Ralph,
thanks for the response, 
Integrating it with existing commands would make sense, but I also use this workflow (using my own fork of ESPlorer) frequently in a line-by line method as a poor-mans step-by-step debugger. 
There just running line by line without the need to select the line actually does make sense.

RalphHogenbirk, 2018-10-06:
I didn't think of that use case yet. For that it will be worth an extra command. I still might like to integrate the multi-line selection feature in the current run button as well, just because we have this button visible in the UI and people know how to use it already. We'll also keep the new command like you made it, which will then have both the selection and the single-line feature. 
I'll think about it a bit more on monday.

Josverl, 2018-10-11:
I took some time to make the additional changes wrt Focus and Indentation.

Note that I did not implement the integration with the Run Command, as @RalphHogenbirk was still thinking about that one.

Let me know what you think, and happy to hear from tests on other platforms.

/Jos

sebastienkrut, 2018-12-19:
Wouldn't it be good to give the focus to the next line of code when a line (or a block) of code has been executed?
Doing so would allow to run few lines in a sequence (even empty lines).

Josverl, 2018-12-27:
Hi Sebi , thanks for the suggestion.
I think that does make sense but not in all cases .
the code currently differentiates between : 
1. some text/code selected ( part of a line , or multiple lines )  - run just the selection 
2. nothing selected --> runs the line the cursor is on 

for 2. it would make sense to move the cursor down to the next line (if any) 
for 1. the logic to do that would be quite hard , the cursor might be at the top of a multiline selection , or the user may have selected just a part of a statement   , ie just selected  `position` or  `positon <= limit `  in : 

    if positon <= limit : 
        dostuff()

in that case (1.) I do not think it would make sense to move down the cursor 
what do you think ?

Josverl, 2018-12-27:
> I didn't think of that use case yet. For that it will be worth an extra command. I still might like to integrate the multi-line selection feature in the current run button as well, just because we [...]

@RalphHogenbirk , 
sorry for the long silence, however I have spent some additional time at debugging and fixing some of the quirks.  some of it was a typical case of 'it works on my machine' which I think was caused by me making edits to the files in the /lib folders while debugging , and not in the /src folder. 

if you have tips how to avoid needing to manually navigate & correlate between /src and /lib files that would be welcome

sebastienkrut, 2018-12-29:
@Josverl: I agree with you, it doesn't make sense to go to the next line in case 1.
It could be handy for case 2 still.

RalphHogenbirk, 2019-01-02:
I think it's a good idea (for case 2 only), but it's not expected default behaviour. I'd say make it a configuration option that defaults to false. We shouldn't assume people run the next line after this one, if they are using the run-by-line feature. I can think of a lot of cases where it's unwanted behaviour to skip to the next line.

Josverl, 2019-01-08:
Hi Ralph, 
I have made a series of further improvements to improve the stability. 
as part of this have also : 

* fixed the incurred line# reporting (no GIThub issue) 
* made some updates to the included upy code so it runs on more boards or more stable ( GetFreeSpace , WiFi SSID , file download / upload code) 
* commented relevant sections of code as per my understanding 
* created a [diagram ](https://github.com/Josverl/pymakr-vsc/blob/improve_runselection/documentation/pymakr-methods%20and%20additonal%20chars.pdf)showing the dependencies between the different functions 
* have removed a number of functions that never get called in VScode. ( but perhaps are used in Atom ?) 
* I have tested this (manually) against several generic boards on both serial and telnet connections

Note: I have not yet added the LineDown feature as Sebastian requested 
See : https://github.com/pycom/pymakr-vsc/compare/master...Josverl:improve_runselection

I realize this has grown into a quite large PR, or alternatively we could decide break it into a few smaller ones by cherry picking, but that would result in additional testing.

Please let me know how you would like to proceed .

/Jos

RalphHogenbirk, 2019-01-09:
@Josverl That's some proper work! The diagram is crazy :) Definitely shows it's not the best code setup and might need some refactoring at some point.

I'll check out your latest code on monday, maybe later today, and will go over all the changes. I'll give you some feedback so we can tweak it and decide how and where to merge everything to. 

In december I was working on atom mostly, implementing some improvements for uploading big files, using zlib compression (check the [1.5 branch](https://github.com/pycom/pymakr-atom/commits/1.5)). I think the best thing would be to create a new release branch in the VSCode package and merge those upload features and your PR to that. If that all works nicely, we can release that version, but also implement your improvements back into the atom package where needed.

sebastienkrut, 2019-02-19:
Wouldn’t it be neat to use **Drag&Drop** rather than Copy/Paste of line(s) of code between a file and the REPL?

This would improve the user’s experience the same way evaluating line(s) of code by pressing CTRL+ENTER did.

I wonder how complex this is to be implemented and if this feature makes sense?

With kind regards,

Josverl, 2019-02-27:
One of the aims of vscode is to minimize the need to use the mouse,

Have you considered the Ctrl-Shift-Enter option that pymakr now has ?

/Jos

closed 2019-03-11 by Josverl