pycom/pymakr-vsc #21: Pymakr(1.0.3) error from last update vsc (1.27.2)

emptygiiiz, 2018-10-06:
Before update it can work normally after install update, When i start VSC it auto connect to board from address in file pymakr.json warnning show
(Running the contributed command:'pymakr.connect' failed.) i can't command in pymakr.

VS Code version: Code 1.27.2 (f46c4c4, 2018-09-12T16:17:45.060Z)
OS version: Windows_NT x64 10.0.17134
Pymakr version: 1.0.3

RalphHogenbirk, 2018-10-08:
Did you update from 1.27.1 to 1.27.2? On mac these versions work fine together, but 'll do a quick test with that version on windows to try and reproduce it. I suspect another issue with the serialport library, for which I'm building a more permanent solution now. I'll let you know.

sebastienkrut, 2018-12-19:
I am facing a small bug currently (Pymakr 1.0.4, VSCode 1.30.1):
Every time I open VSCode and select the Pycom Console, it appears but remains blank.
Trying to connect or disconnect to a board doesn't help.
The only workaround I have found is to run Pymakr > Help first.

RalphHogenbirk, 2019-05-16:
@sebastienkrut old issue, but was this issue fixed in the latest release? I've not seen any other reports of this issue.

sebastienkrut, 2019-05-22:
Dear Ralph, I am still experiencing this issue sometimes. But Pymakr > Help makes it work.

closed 2019-11-11 by pedroknup