pycom/pymakr-vsc #23: Download/synch function is unable to download/list files reliably

Josverl, 2018-10-14:
## What are the steps to reproduce this issue?

this is an extreme repro, but as i found the download to be unreliable I wanted to narrow down the issue or issues

1. use a board with a pre-existing file structure on the flash 
2. remove / rename the boot.py 
3. reset the board & connect with pymakr
4. try to sync download 

## What happens?
pymakr is unable to download any files.
the reason for this is that `import os`  is missing from the code that is used to list the current files on the board 
in addition the resulting error condition is not detected/handled properly 
-
## What were you expecting to happen?
the download to work robustly 

## Any logs, error output, etc?
After additional additional errorhandling the following error messages become apparent : 
```
[silly] Sync | Entered raw mode
[verbose] Pyboard | Executing code:import ubinascii,sys
list = ubinascii.hexlify(str(os.listdir('/flash/')))
sys.stdout.write(list)
[silly] Pyboard | Executed raw code, now resetting
[info] Pyboard | Soft reset
[silly] Pyboard | Received message: OK
[silly] Pyboard | Buffer length now 2
[silly] Pyboard | Waiting for >
[silly] Pyboard | Received message: Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
NameError: name 'os' is not defined
>
```
## Any other comments?
fixing this specific error is trivial by adding the import to `import ubinascii,sys` --> 'import ubinascii,sys,os'

I would suggest that the download code should is by adding errorhandling and retries as there are additional cases in which errors are returned by the board , and not detected.
as a result the error messages are treaded as base64 encoded strings , and an attempt is made to decode these , and the resulting downloaded file is corrupted or contains garbled information at best. 

also the download code does not close the files, which will/might consume/leak memory on the MCU leading to failed downloads once memory is exhausted.  
I'm not quite sure if the raw repl + reset (Ctrl-A, <leaking code>, CTRL-D will be able to free the memory in this case, but I think its better to avoid the lead than assume that the RAW Repl will cleanup 
 
## What versions of software are you using?

**Operating system:**
`Microsoft Windows [Version 10.0.17763.55]`

**VSCode version:**
```
Version: 1.28.1 (system setup)
Commit: 3368db6750222d319c851f6d90eb619d886e08f5
Date: 2018-10-11T18:13:53.910Z
Electron: 2.0.9
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
Architecture: x64
```
**Pymakr version:**
`1.0.3 `

closed 2018-10-15 by RalphHogenbirk

Josverl, 2018-10-15:
Thanks Ralph,

I'm working on some more improvements to the download stability as I kept running out of memory.
I added GC and refactored the code a little to avoid leaking file handles 

https://github.com/Josverl/pymakr-vsc/commit/f19cf18224455e28c9a6c173769f0e6ef35cd700 

It's not fully baked yet, I'll send a PR when I think it is.

/Jos

RalphHogenbirk, 2018-10-15:
Thanks for helping out @Josverl! Very much appreciated! 

Hint for running out of memory: there is a setting that puts the board in safe-boot before every upload or download, which should help with the memory issue. But reducing memory usage is always a good thing of course. Will keep an eye out for the PR.

PR #24 (2018-10-15): Fix missing import in list_files
## What does this implement/fix? Explain your changes.
add a missing `import uos as os`  to the micropython code to generate a filelist.

## Does this close any currently open issues?
yes, this fixes #23 

## Any other comments?
this is just a fix for the missing import, no additional error checking has been added

## Where has this been tested?

**Operating system:**
`Microsoft Windows [Version 10.0.17763.55]`

**VSCode version:**
```
Version: 1.28.1 (system setup)
Commit: 3368db6750222d319c851f6d90eb619d886e08f5
Date: 2018-10-11T18:13:53.910Z
Electron: 2.0.9
 [... cut]