pycom/pymakr-vsc #25: UX does not always show the correct Connected/Disconnected or Running state

Josverl, 2019-01-08:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. connect to a board 
2. upload and run a failing script
3.  manually reset the board 

## What happens?

* Ux still shows Connected and Running 
* when you disconnect and reconnect , the connecttion state is fixed , 
** however even then the Run state still shows running 

## What were you expecting to happen?

- at best : the running and connect state to automatically detect the restart 
- at minumon : the run state to be stopped / reset on connect 

## What versions of software are you using?

**Operating system:**
Microsoft Windows [Version 10.0.17763.195]

**VSCode version:**
Version: 1.30.2 (user setup)
Commit: 61122f88f0bf01e2ac16bdb9e1bc4571755f5bd8
Date: 2019-01-07T22:54:13.295Z
Electron: 2.0.12
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
OS: Windows_NT x64 10.0.17763

**Pymakr version:**
1.0.4

RalphHogenbirk, 2019-01-09:
@Josverl thanks for reporting. Question: are you connecting over serial or over wifi in this case? And can you post an example failing script that you use to reproduce it?

Josverl, 2019-01-12:
@RalphHogenbirk , connecting over serial , and I now I see that I never see the connected state on my machine with the 1.0.4 release as you see below. 

[image]

( starts connected , then disconnect , then connect again )

I think I found and solved the cause for this in : da03f9e7a5241db1bf36da903675229dfbbf3f51
but please review that

RalphHogenbirk, 2019-01-14:
@Josverl I can't reproduce it on mac, but I can test if your commit still works? Can you tell me on which branch this fix commit (da03f9e) is on? For some reason Github doesn't tell me that.

RalphHogenbirk, 2019-01-15:
@Josverl I've merged your latest pull request, the latest code from the 1.5 branch of pymakr-atom and your da03f9e commit to the current [develop branch.](https://github.com/pycom/pymakr-vsc/commits/develop). Still working on implementing and testing some of the new features from atom, but I'll release this soon. Let me know if this version has a working connection state on your machine!

Josverl, 2019-03-28:
I think this is now fixed by a combo of your changed and #33

closed 2019-03-28 by Josverl