pycom/pymakr-vsc #27: Why Extension limited to PYCOM only ?

timpur, 2019-02-03:
# Question, Why only support PYCOM boards ?

im interested to by pycom isnt more open to helping micropython generally ? As it get better so does pycom ? Maybe im missing an article about this? Thanks

https://github.com/pycom/pymakr-vsc/blob/f54ff1798aabda765ffd8e8450c1539487b62334/src/connections/pyserial.js#L50

timpur, 2019-02-03:
My bad. Thank you for that information. 

Im using an Nodemcu esp8266 and the FTDI chip has a weird brand, so thats why the atoconnect doesnt pick it up.

i wonder if there is a better way of detecting micropython ?

Josverl, 2019-02-08:
Currently autodetect is based on matching the serial port description to a hardcoded list of names.
i had previously been thinking to allow that to be extended via the global settings to include some of the generic CP210x variants, 

However i've not got around to doing that , also as im battling a bluescreen that is caused by that same driver  🤔

timpur, 2019-02-09:
If I have time I'd like to extends the list to be global :) As you've said.

closed 2019-02-09 by timpur

RalphHogenbirk, 2019-02-11:
Agreed on this, the autoconnect list should definitely be in global settings! Should be easy to implement, might do this later today or tomorrow!

RalphHogenbirk, 2019-02-13:
@timpur @Josverl Implemented on the develop branch in commit eb90379. Let me know if this is what you guys had in mind too.

timpur, 2019-02-14:
Looks good I'll try it out later :)