pycom/pymakr-vsc #28: [bug] VSCode update 1.31.0 breaks pymakr addin

Josverl, 2019-02-08:
Issue Type: <b>Bug</b>

Start : working environment with pymakr
>  VSCode updates to 1.31.0 

After update pymakr addin does not work anymore 
any attempt to use results in an error such as: 
`command 'pymakr.listCommands' not found`

Tried : 
- re-installation of add-in does not resolve the issue 

see also : https://github.com/Microsoft/vscode/issues/68230

Extension version: 1.0.5
VS Code version: Code 1.31.0 (7c66f58312b48ed8ca4e387ebd9ffe9605332caa, 2019-02-05T22:35:56.624Z)
OS version: Windows_NT x64 10.0.17763

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz (8 x 2112)|
|GPU Status|2d_canvas: enabled<br>checker_imaging: disabled_off<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>native_gpu_memory_buffers: disabled_software<br>rasterization: enabled<br>surface_synchronization: enabled_on<br>video_decode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Memory (System)|15.93GB (6.54GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|

</details>

Josverl, 2019-02-08:
VSCode suggests that this is _very likely causes by the Electron update_

curtis-hendrix, 2019-02-08:
Cross - posting a workaround to at least get Pymakr functioning again, if you can't wait for Microsoft and/or Pycom to fix this problem:

    1. Uninstall Visual Studio Code
    2. Delete the `Code` folder from `%appdata%`.
    3. Delete the `.vscode` folder from `%userprofile%`
    4. Download Visual Studio Code version 1.30 from here: https://code.visualstudio.com/updates/v1_30
    5. Install Visual Studio Code.  **DO NOT LAUNCH VISUAL STUDIO CODE AFTER INSTALLATION!!!**
    6. Open `%appdata%`.  Create a new folder called `Code`.  Within `Code`, create a new folder called `User`.  Within `User`, create `settings.json` file. Add the following to `settings.json`:
       {
       "update.channel": "none"
       }

This will disable automatic updates.
### **You must change the `update.channel` settings BEFORE launching VSC! Otherwise VSC will update when you close it, _even_ if you change the setting within VSC!**

Josverl, 2019-02-10:
```
Activating extension `pycom.pymakr` failed: The module '\\?\C:\develop\pymakr-vsc\precompiles\serialport-win\build\Release\serialport.node'
extensionHostProcess.js:331
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

57 -->  NodeJS 8.x.x aka Carbon 
64 -->  NodeJS 10/11 

https://nodejs.org/en/download/releases/

RalphHogenbirk, 2019-02-11:
V1.0.6 has been released with the fix for this. Had to recompile the included serialport lib builds to fit the electron version of vscode.

AndyKassell, 2019-02-11:
So now in vscode  the Pycom Console terminal window opens as expected (great thanks) but cannot connect to pycom module,  I get

> Autoconnect: No PyCom boards found on USB

however when I use the list serial ports command It lists the port with the pycom attached
 
> Found 1 serialport
> COM5 (Microsoft) (copied to clipboard)

if I fire up Atom it connects fine to the pycom on COM5 and enters REPL as extected.

RalphHogenbirk, 2019-02-11:
@AndyKassell good to hear hat's working so far.

The connect issue is related to how autoconnect works. It only autoconnects to boards that are recognized as possible pycom boards. These are devices that report with manufacturer Pycom, Pycom Ltd. or FTDI. Your's reports as 'Microsoft' (I guess because you use your own FTDI converter of that brand?) which is not attempted to connect to. So for now, you can't use the autoconnect feature with that specific hardware.

We're planning to make that list into a configuration param, so you could add 'Microsoft' to the list and solve your issue. (It's discussed in issue #27)

AndyKassell, 2019-02-11:
@RalphHogenbirk Thanks for the quick reply. Not using any 3rd part serial board, it's an expansion board 3.1 connected directly to USB port, I have tested my pysense and pytrack base boards which also show as "Microsoft", but as I sadi they all autocennect using Atom, just not VsCode if that help at all.

RalphHogenbirk, 2019-02-11:
Ah you're right, on windows the expansion board 3.1 apparently shows up as 'Microsoft'. I wasn't aware of this. I'll add this to the default list and will also add the option to add other comports through the settings file.

For now you could hack it in if you wanted to. There's an array here:
`~/.vscode/extensions/pymakr/pycom.pymakr-1.0.6/lib/connections/pyserial.js:172`
If you add "Microsoft" to that list, autoconnect should work for you.

AndyKassell, 2019-02-11:
Hi @RalphHogenbirk, Atom works as supported list in pymakr-atom pyserial.js is as follows

`static COMPORT_MANUFACTURERS = ['Pycom','Pycom Ltd.','FTDI','Microsoft','Microchip Technology, Inc.']`

where vscode pyserial.js is

```
  static COMPORT_MANUFACTURERS(){
    return ['Pycom','Pycom Ltd.','FTDI']
  }
```

and does not contain MS etc.. I did try to build the extension locally so I could modify but ran into issues and had to bail due to time constraints. But currently all my directly connected pycom boards are showing as Microsoft in the usb driver 

WIll use Atom for now.

RalphHogenbirk, 2019-02-11:
Ok, that indeed explains the different behaviour in atom, where the list is already complete.

If you don't have time to build the extension locally, just change it in the lib folder of your extension (location in my last comment) and it'll at least work until the next update.

AndyKassell, 2019-02-11:
@RalphHogenbirk Chnaged in lib and that has worked, many thanks for you help.

Josverl, 2019-02-13:
@ralph , thanks for the fix. I tried to figure out how to rebuild, or download the precompiled myself but could not figure out how to make that work.

that way it would be possible to prevent a similar issue for occurring in the with the immediate need for a re-compile
there seems to be an option to 
- detect the VSCode node module version ( ie current 64 ) 
- get the serialport bindings for future node versions as published on : 
https://github.com/node-serialport/node-serialport/releases/tag/%40serialport%2Fbindings%407.1.4
- store these bindings 
- and then load the relevant bindings for the platform + Module Version 

ie something like : 
```
const SerialPort = require('@serialport/stream')
const Binding = require('@serialport/bindings_<PLATFORM>_<MODULE_VERSION>')
const parsers = require('./parsers')
/**
 * @type {AbstractBinding}
 */
SerialPort.Binding = Binding
/**
 * @type {Parsers}
 */
SerialPort.parsers = parsers
module.exports = SerialPort

```

However lacking understanding on how you currently create the bindings , i failed to get this working.
Note that it may also allow reducing the amount on included source code in the current package, which seems to have a significant influence on the installation time.
let me know if you think that is a worthwhile persuing

closed 2019-02-13 by Josverl

RalphHogenbirk, 2019-02-13:
@Josverl nice work. Just to give you an idea of how I rebuild the serialport precompile right now, I made a script (scripts/rebuild.js) to do this. It basically:
- Removes node_modules
- Runs an apm install
- Apm install triggers scripts/install.js
   - This installs serialport 
   - Rebuilds node_modules/serialport for electron using electron-rebuild
- Removes old /precompiles/serialport-<os> folder
- Moves node_modules/serialport to precompiles/serialport-<os>

Lately, serialport added dependancy grouping, which broke the end result of this. My quick fix was to remove al references to the @serialport/ folder from the precompiled version, and everything works. Very likely not the best solution, but I have yet to figure out how this works exactly.

I like your suggestion for just downloading the bindings for each platform instead of manually building it. That would save a lot of troubles and installation time.