pycom/pymakr-vsc #29: MCU explorer

Josverl, 2019-02-13:
# Feature Request 🚀

Add a file explorer t show the files on the devboard 

Based on : \vscode-extension-samples\tree-view-sample

Under the listed folder 
	• / root fs , may be empty) 
	• /flash 
	• /sd (if mounted) 

Primitives 
	- ListDir -recursive 
	- Upload single File 
	- Download Single File 
	- Delete file 
	- Rename file
	- Copy file (on MCU )
	- Run file on board (exec( foo.py ) 
	- Import file on board (import foo
	- View File (download & view )
Actions / use cases : 
- Refresh on initial connect 
- Refresh on demand 
- Refresh on upload / synch
- Refresh after run (optional) 
- Auto Upload a file after local Save (using file filters) 

## Solution Approach 

* There already is a sample of a VSCode extension that renders a treeview of files. this can be adopted 
* a new I/ improved methods is needed to list the files on the board as the current implementation of list files does not appear to be able to reliably list the files on the board. For this I already have a PoC python scriptlet that returns all files and folder and optionally the CRC32 checksum as a json object 

this would also allow a lot of rather complex code flows to list / traverse folders to be replaced by a much simpler approach.
to only downside that I can see is that the script and generating the JSON will have somewhat larger memory requirements than the current approach.

RalphHogenbirk, 2019-02-13:
Wonderful idea! I didn't know about this specific API to control a tree view, but it seems like a relatively easy implementation. I'll have a look at your code very soon.
In most cases people will just upload files to the board and just trust that they are there and working well. But I know in some use cases you have to fall back to FTP to check on logs and so on. This case is also kind of covered by the download button, but it's kind of a blind process. So it's nice to have the option to just quickly check what's actually on the board and download or upload a single file. I also like the 'upload file on save' idea, which is also implementable for atom (even without a tree view)

Right now we're making a new layout, design and feature list for Pymakr (focused on Atom for now) and this idea kind of fits those plans. We will likely publish some of those plans later today / tomorrow and would love to have your feedback on it too.

Josverl, 2022-04-08:
This is is planned to be covered in V2 , although there are still a few bugs open  #205
@jakobrosenberg , do you want to keep this open , closed ?

jakobrosenberg, 2022-05-24:
Sorry completely overlooked this. Let's close it as we now have the feature.

closed 2022-05-24 by jakobrosenberg