pycom/pymakr-vsc #30: VScode gets stuck in "Installing" state

arthurlutz, 2019-02-13:
## What are the steps to reproduce this issue?

1. Install / upgrade the pymakr extension in vscode or via the command line
2. Remove extension using `vscode --uninstall-extension pycom.pymakr`
3. Try again, `vscode --install-extension pycom.pymakr --verbose`

## What happens?

VSCode gets stuck in the "Installing" state. No verbose logs.

## What were you expecting to happen?

installing the extension 

## Any logs, error output, etc?

Nope

## Any other comments?

This might be related to #28 

## What versions of software are you using?

**Operating system:** Ubuntu bionic

**VSCode version:** 1.31.0

```
snap list 
[snip]
vscode                 1.31.0-1549443364       77    beta      snapcrafters  classic
```

**Pymakr version:** 1.0.6

RalphHogenbirk, 2019-02-13:
@arthurlutz thanks for the report. plugin was compiled and tested on ubuntu 16.04, so I suspect it might be specific to ubuntu bionic. Looks like it's time to update my virtual machine to 18.04 :)

arthurlutz, 2019-02-13:
@RalphHogenbirk thanks for taking a look at this.

```
$ nodejs  --version
v8.10.0
```
Is that a problem ? 

Do you have an idea of a workaround ?

RalphHogenbirk, 2019-02-13:
V8.10 should be fine, i'm running v8.4.0 myself.

Not sure about a workaround for now, perhaps you could setup the development environment ([manual in the readme](https://github.com/pycom/pymakr-vsc#developing)) and run it like that for now. There's also Pymakr for Atom as an alternative.

RalphHogenbirk, 2019-05-16:
@arthurlutz Has this issue been solved in the latest release?

closed 2019-12-02 by pedroknup