pycom/pymakr-vsc #32: Unable to upload files on M5 Stack using VS code. Possibly a conflict with Pymakr

prateekchikara, 2019-03-11:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. I am working on a weather station code using Micropython on VS code.  What I observed that while uploading the files on M5Stack using a VS code, it gets stuck on the files which has code connecting to the wi-fi. For example in weather station the main file actual size is 2KB however it shows 0KB in VS code during upload to M5 Stack. I have already tried with 3 different M5Stack devices and on 2 separate computers as well. In order to be sure I also tried uploading different files as well. The files without the wi-fi folders were already uploaded however any file which required a wi-fi connectivity, say connecting it to the hotspot it gets stuck while uploading. I am attaching the error log I got under "toggle developer tools" in this email. I am not sure if it is the pymakr version that is creating the issue or something else. However, it used to work before say last month and then all of a sudden the upload stopped. I've even tried resetting my computer however it was of no avail. Since it was working before in January, I've tried with the older version of VS code but was of no avail. So, far what I think it maybe a configuration issue with the pymakr plugin after a certain update however, since my knowledge is on a beginner scale I am not certain. I have already 

## What happens?

What happens is the upload gets stuck and it doesn't continue. Reconnecting and rebooting the device doesn't help either.

## What were you expecting to happen?

I was expecting the upload to finish and then the weather station device(M5 Stack)
 to work

## Any logs, error output, etc?
https://gist.github.com/prateekchikara/037203fb42c6201bfc07a7f4fa0e53f1

## Any other comments?

## What versions of software are you using?

**Operating system:** 

Windows 10

**VSCode version:**

1.32.1

**Pymakr version:**

1.0.7

prateekchikara, 2019-03-11:
[image]
 I am also attaching the VS code screenshot

RalphHogenbirk, 2019-03-11:
Hi, thanks for the extended report!

First, i've spotted a small bug in some error handling code of Pymakr, which causes the upload to freeze instead of giving a proper error message and retrying to upload that specific file. You could hot-fix it if you want, by editing line 51 of pycom.pymakr-1.0.6/lib/board/shell-workers.js. The 'var ' declaration is missing for the 'err_mssg' variable (see commit https://github.com/pycom/pymakr-vsc/commit/16797016c915519dc293425e93614615d0d207ce)

Now, I also notice you're using 1.0.6, try switching to 1.0.7 before editing the above line.

This doesn't explain why your file fails to upload in the first place, but the fix above might give you some more information to debug it. Maybe your device is full?

Let me know what you find.

Josverl, 2019-03-11:
Hi Prateek, Ralph,

There seem to be a few issues, indeed the error message, but that did mostly hide the underlying error.
The current code is leaking a filehandle in the get_hash function
, and at some point micropython runs out of filehandles.

I guess on pycom boards this is not as obvious, as the safeboot is also a workaround to freeing filehandles and memory; but the codefix is quite pythoneque;
by using:

```
import uhashlib,ubinascii,sys
hash = uhashlib.sha256()
with open('yourfilehere.py', 'rb') as f:
  while True:
    c = f.read(512)
    if not c:
        break
    hash.update(c)
sys.stdout.write(ubinascii.hexlify(hash.digest()))
```

The same may happen in the compress function ( cool ; need to figure out how to turn this on ..) to I suggest updating that code as well.

Ill send a PR shortly.

RalphHogenbirk, 2019-05-16:
@prateekchikara just to come back to this, have the fixes in the latest release solved your issue?

jakobrosenberg, 2022-05-24:
Closing stale issues. If you think this issue should stay open, let me know and I'll reopen it.

closed 2022-05-24 by jakobrosenberg