pycom/pymakr-vsc #34: Development environment \incorrect dependencies

Josverl, 2019-03-11:
## The steps to reproduce this issue?

1. On windows 10
2. Clone the repo 
2. npm install --save-dev babel-core

## What happens?

C:\develop\repro\pymakr-vsc>npm install --save-dev babel-core
npm ERR! code ETARGET
npm ERR! notarget No matching version found for parser-regex@^1.0.5
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.
npm ERR! notarget
npm ERR! notarget It was specified as a dependency of 'serialport'
npm ERR! notarget

[image]

An indeed there is no 1.0.5 version on NPM 

npm view parser-regex 
shows that the latest: 1.0.2

A similar error is shown in my regular environment.

## What were you expecting to happen?
environment to work 

## Any other comments?
This may have to do with different version of the serial library, as I tried to experiment with multiple version of that a while ago. I assumed I cleaned that all up , but there may be stray libraries.

## What versions of software are you using?

**Operating system:**

**VSCode version:**
n/a

**Pymakr version:**
https://github.com/pycom/pymakr-vsc.git

Josverl, 2019-03-12:
it appears that the issue stems from the serial port packages v 1.0.5 listed in the package-lock.json, 
I can only find 1.0.2 on npm 

```json
{
  "name": "pymakr",
  "version": "1.0.5",
  "lockfileVersion": 1,
  "requires": true,
  "dependencies": {
    "parser-byte-length": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-byte-length/-/parser-byte-length-1.0.5.tgz",
      "integrity": "sha512-GCz/v/KG2Wv7SdQ2nv8jYGBY6D4h5tibj9bs0+pnryCDAr8xmmvnesFW15FIu4rwOMgsKhCHyp7roD8bRGs63A==",
      "requires": {
        "safe-buffer": "^5.1.1"
      }
    },
    "parser-cctalk": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-cctalk/-/parser-cctalk-1.0.5.tgz",
      "integrity": "sha512-VdoG1rRXb5deHM1c9Akn9djoJuHn030v7owYHEqpJeS6Rs6wrC4Hrkw8NxvV9ZPlMqAJ+5uJCaAUzB1tbVd3rA==",
      "requires": {
        "safe-buffer": "^5.1.1"
      }
    },
    "parser-delimiter": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-delimiter/-/parser-delimiter-1.0.5.tgz",
      "integrity": "sha512-srDzeNwGM/GjtqK/nFDRIDpcZ6XDgkakFMXBtNDSI+XP6fqO1ynEZok8ljKJxM2ay0CNG83C6/X2xIOHvWhFYQ==",
      "requires": {
        "safe-buffer": "^5.1.1"
      }
    },
    "parser-readline": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-readline/-/parser-readline-1.0.5.tgz",
      "integrity": "sha512-QkZoCQPHwdZOMQk7SHz3QSp7xqK4jdNql9M80oXqWt7kNhFvNXguWzf17FfQrPRIb0qiz+96+P6uAOIi02Yxbg==",
      "requires": {
        "parser-delimiter": "^1.0.5",
        "safe-buffer": "^5.1.1"
      }
    },
    "parser-ready": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-ready/-/parser-ready-1.0.5.tgz",
      "integrity": "sha512-U/ZkxyY35Z7WrDc0O8TGcGPOdwv6fGVJcZq5vXVko2MRt8wiKVD192mmbfTRZXFAX+rARXtQa3ad3yJzXVhb1g==",
      "requires": {
        "safe-buffer": "^5.1.1"
      }
    },
    "parser-regex": {
      "version": "1.0.5",
      "resolved": "https://registry.npmjs.org/parser-regex/-/parser-regex-1.0.5.tgz",
      "integrity": "sha512-sX3tRuwwwGV+CZbKEUAKZD/wtG8ZRcGxbiDIm8nyzsPCGv52ck3RlQ9Vp4K8fYjcrGGwm3BWizC4uSzaTLOk1A==",
      "requires": {
        "safe-buffer": "^5.1.1"
      }
    },
    "@types/mocha": {

```

removing the lock file and running `npm install` seems to fix the issue.

Is this possibly related to the recent rebuild of the serial libraries ?

RalphHogenbirk, 2019-03-20:
Same issue on OSx. Not sure why serialport would refer to non-existent versions of these dependancies, but it sure seems to do just that. I suspect that it's also because we're using an 8 month old version of serialport (6.2.2). 

I suspect this will also be fixed if we update to a never version of serialport, which might be the best way to do it. I do remember that causing some issues in Pymakr (which is why it's still not updated). I'll give it a quick try again and see if it fixes things.

Josverl, 2019-04-01:
Hi Ralph, just some feedback 
Ive been using your version of the serialport 7.1 branch for the last week, and have not seen any serial issues thus far, so that works for me.

RalphHogenbirk, 2019-04-01:
Very good to hear @Josverl, thanks for testing that. Will do a release this week, then we'll see if it holds up with all users.

closed 2019-05-16 by RalphHogenbirk