pycom/pymakr-vsc #36: Connects to wrong COM-Port and can't choose other port

Lonefish, 2019-03-21:
## What are the steps to reproduce this issue?
1.
2.
3.

## What happens?
I'm trying to connect to my Pycom while there's also another COM device connected. Following [this](https://docs.pycom.io/pymakr/installation/vscode.html) tutorial I pasted 'COM20' into the address in pymakr.json, but when I run 'Pymakr > Connect' it tries to connect on 'COM24'

## What were you expecting to happen?
Connecting op COM20

## Any logs, error output, etc?
//Pymakr > List serial ports:
Found 3 serialports
COM24 (Microsoft) (copied to clipboard)
COM20 (Microsoft)
COM1 ((Standaardpoorttypen))

// Pymakr > Connect
Connecting to COM24...
> Failed to connect (Error: Port is not open). Click here to try again.
Connecting to COM24...
> Failed to connect (Error: Port is not open). Click here to try again.

pymakr.json
{
	"address": "COM20",
	"username": "micro",
	"password": "python",
	"sync_folder": "",
	"open_on_start": true,
	"safe_boot_on_upload": false,
	"py_ignore": [],
	"fast_upload": false,
	"sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
	"ctrl_c_on_connect": false,
	"sync_all_file_types": false,
	"auto_connect": true,
	"autoconnect_comport_manufacturers": [
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc."
	]
}
## Any other comments?
Deleting "Microsoft" from autoconnect_comport_manufacturers makes it not connect to COM24 anymore, but doesn't let me connect to COM20 either. It seems that the address parameter doesn't work at all.

## What versions of software are you using?
//

**Operating system:**
Windows 10

**VSCode version:**
Version: 1.32.3 (system setup)
Commit: a3db5be9b5c6ba46bb7555ec5d60178ecc2eaae4
Date: 2019-03-14T23:43:35.476Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 10.0.17134
**Pymakr version:**
1.0.7

Josverl, 2019-03-21:
Hi Lonefish,

I suspect that your 'Project > Settings' are set to port 24, 
which would override your 'Global > Settings' that you have set to 20.

I treat the Globals as a default and the Project as a per project override.
although there are a few settings that only work on global level.

/jos

Lonefish, 2019-03-22:
Somehow it seemed to have fixed itself the day after.. COM20 is filled in in the project settings without me putting it there (I closed VSCode after this issue and just reopened it now). Apparently a PEBCAK problem, my apologies :)

closed 2019-03-22 by Lonefish