pycom/pymakr-vsc #41: Not having node installed leads to unusable IDE

jimbobbennett, 2019-04-27:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Install the Pymakr extension without Node installed

## What happens?
A constant stream of error popups that are too fast to see what they say and render VS Code unusable
Fix is to install Node, but no obvious indication this should happen

## What were you expecting to happen?
VS Code should be usable, and maybe a single error telling you to install Node.
Ideally give the option for an auto install of Node

## What versions of software are you using?

**Operating system:**
Windows 10

**VSCode version:**
1.33.1

**Pymakr version:**
1.0.7

RalphHogenbirk, 2019-04-27:
@jimbobbennett This is indeed a bug, but luckily it's already solved in the develop branch. Current release shows an error message but continues loading, which results in the terminal crashing. Pymakr tries to reload the terminal, causing an infinite loop, drowning the initial error message.

With the fix, it blocks the entire plugin from loading and just shows the error message. The plan is to release a new version on Monday that includes this fix.

RalphHogenbirk, 2019-05-16:
Solved in the latest release

closed 2019-05-16 by RalphHogenbirk