pycom/pymakr-vsc #42: Pymakr does not load on Visual Studio Code Start

curtis-hendrix, 2019-04-30:
I made the critical mistake of shutting down Visual Studio Code earlier today and now I cannot get the Pymakr extension to load on Code startup.  I have tried all 5 of the available versions for Windows (User 32 / 64 bit, System 32 / 64 bit, and the zip folder), performing a complete wipe of Code between each installation ([instructions here](https://stackoverflow.com/questions/49166530/how-can-i-do-clean-uninstall-so-no-settings-extension-etc-is-left)) 

I see in the console output (available under Developer Tools) that Pymakr is failing due to `pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node is not a valid Win32 application`

I'm not sure what (if anything) changed on my system earlier today.  All I did was shut down Visual Studio Code so I could run the Pycom firmware updater on a GPy, and then Pymakr refused to load the next time I started Visual Studio Code.

# Visual Studio Code information
Version: 1.33.1
Commit: 51b0b28134d51361cf996d2f0a1c698247aeabd8
Date: 2019-04-11T08:27:14.102Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 10.0.17134

Output from the Console window (available under Help-> Toggle Developer Tools):

`extensionService.ts:800 Activating extension 'pycom.pymakr' failed: \\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node is not a valid Win32 application.
\\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node.
_logMessageInConsole @ extensionService.ts:800
console.ts:134 [Extension Host] Activating extension `pycom.pymakr` failed:  \\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node is not a valid Win32 application.
\\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node
t.log @ console.ts:134
console.ts:134 [Extension Host] Here is the error stack:  Error: \\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node is not a valid Win32 application.
\\?\c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node
	at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
	at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
	at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
	at Module.load (internal/modules/cjs/loader.js:620:32)
	at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
	at Function.Module._load (internal/modules/cjs/loader.js:551:3)
	at Function.t._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:729:537)
	at Function.t.getExtensionPathIndex.then.a._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:691:639)
	at Function.t.getExtensionPathIndex.then.r._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:655:197)
	at Module.require (internal/modules/cjs/loader.js:658:17)
	at n (c:\VSCode\resources\app\out\vs\loader.js:15:874)
	at bindings (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\bindings\bindings.js:81:44)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\lib\win32.js:1:191)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\lib\win32.js:111:3)
	at Module.t._initNodeRequire.r._compile (c:\VSCode\resources\app\out\vs\loader.js:16:90)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
	at Module.load (internal/modules/cjs/loader.js:620:32)
	at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
	at Function.Module._load (internal/modules/cjs/loader.js:551:3)
	at Function.t._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:729:537)
	at Function.t.getExtensionPathIndex.then.a._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:691:639)
	at Function.t.getExtensionPathIndex.then.r._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:655:197)
	at Module.require (internal/modules/cjs/loader.js:658:17)
	at n (c:\VSCode\resources\app\out\vs\loader.js:15:874)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\lib\index.js:6:22)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\lib\index.js:17:3)
	at Module.t._initNodeRequire.r._compile (c:\VSCode\resources\app\out\vs\loader.js:16:90)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
	at Module.load (internal/modules/cjs/loader.js:620:32)
	at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
	at Function.Module._load (internal/modules/cjs/loader.js:551:3)
	at Function.t._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:729:537)
	at Function.t.getExtensionPathIndex.then.a._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:691:639)
	at Function.t.getExtensionPathIndex.then.r._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:655:197)
	at Module.require (internal/modules/cjs/loader.js:658:17)
	at n (c:\VSCode\resources\app\out\vs\loader.js:15:874)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\serialport\lib\index.js:2:17)
	at Object.<anonymous> (c:\VSCode\data\extensions\pycom.pymakr-1.1.0\node_modules\serialport\lib\index.js:17:3)
	at Module.t._initNodeRequire.r._compile (c:\VSCode\resources\app\out\vs\loader.js:16:90)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
	at Module.load (internal/modules/cjs/loader.js:620:32)
	at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
	at Function.Module._load (internal/modules/cjs/loader.js:551:3)
	at Function.t._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:729:537)
[... 31 lines cut ...]
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
	at Module.load (internal/modules/cjs/loader.js:620:32)
	at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
	at Function.Module._load (internal/modules/cjs/loader.js:551:3)
	at Function.t._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:729:537)
	at Function.t.getExtensionPathIndex.then.a._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:691:639)
	at Function.t.getExtensionPathIndex.then.r._load (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:655:197)
	at Module.require (internal/modules/cjs/loader.js:658:17)
	at require (internal/modules/cjs/helpers.js:20:18)
	at Function.i [as __$__nodeRequire] (c:\VSCode\resources\app\out\vs\loader.js:31:542)
	at c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:717:918
	at E._doActivateExtension (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:718:2)
	at E._activateExtension (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:716:772)
	at E.<anonymous> (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:714:224)
	at Generator.next (<anonymous>)
	at n (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:11:176)
	at new Promise (<anonymous>)
	at n (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:10:975)
	at Object.actualActivateExtension (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:713:946)
	at p._activateExtension (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:457:765)
	at define._activateExtensions.i.length.Promise.all.o.map.e (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:457:449)
	at Array.map (<anonymous>)
	at p._activateExtensions (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:457:437)
	at p.activateById (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:455:604)
	at E._activateById (c:\VSCode\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:715:497)`

RalphHogenbirk, 2019-05-01:
Hi @curtis-hendrix, sorry to hear this. In the last release (v1.1.0) we've updated the bindings.node file that is now giving you an issue. So either this file is not correct, or the installation scripts are is not placing it in the correct folder somehow on your system.

Could you give me one piece of info: the size in kb of the bindings.node file. (pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node) This will tell me if it was build properly. 

After that you could try to overwrite this file with the one in pycom.pymakr-1.1.0\precompiles\serialport-win32\bindings.node (or serialport-win64 if you're running 64 bit) and then restart vscode.

If that doesn't help and you need it working quickly, here's the previous version in vsix form: https://www.dropbox.com/s/41td19stzgk8fn2/pymakr-1.0.7.vsix?dl=0. Uninstall pymakr, then use 'install from vsix' and select the file.

JavierNR, 2019-05-01:
Hello @RalphHogenbirk @curtis-hendrix 
I have exactly same problem here, my bindings.node file size is:
66,8 KB (68.452 bytes)
Bindings file under precompiles folder is:
160 KB (164.352 bytes)

Replacing the file doesn't solve the problem, will try to downgrade from vsix file

RalphHogenbirk, 2019-05-01:
@JavierNR thanks for tetsing that, sounds like the bindings.node precompile is actually incorrectly build. I'll do some testing on my windows machine later today. Somehow it worked on mine while testing this release with that same build file, but it might be version related. I'll look into this today

curtis-hendrix, 2019-05-01:
@RalphHogenbirk 

> Hi @curtis-hendrix, sorry to hear this. In the last release (v1.1.0) we've updated the bindings.node file that is now giving you an issue. So either this file is not correct, or the installation [...]

My bindings.node file under pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release is 66.8KB (68,452 bytes).  

I copied the bindings.node file from the serialport-win folder into the Release folder, overwriting the original, and now it looks like Pymakr is back in business.  @JavierNR Try both of the bindings.node files.  One's in the \precompiles\serialport-win folder, the other is in \precompiles\serialport-win32.  Also, I'm using the standalone version of VSC, with all of the extentions installed within the VSC folder.

RalphHogenbirk, 2019-05-01:
@curtis-hendrix could you check one thing for me if you have time? Can you run 'node scripts/install.js' from your 'pycom.pymakr-1.1.0' folder and see if any errors pop up? And if not, does pymakr work after running it? Output should be:

'Copy bindings file
Installing electron rebuild
Copy completed
Rebuilding for electron 3.1.6...
done'

curtis-hendrix, 2019-05-01:
@RalphHogenbirk I don't have a `nodescripts` folder in `pycom.pymakr-1.1.0`, but I do have a `scripts` folder with `install.js` inside of it.  I ran it with Windows Script Host, and got this error:

[image]

RalphHogenbirk, 2019-05-01:
@curtis-hendrix the command I typed above was on a line break, that made it confusing. There is a space between 'node' and 'scripts' :) So it's using the 'node' command to run the script, from your command line. Make sure to run it from the pycom.pymakr-1.1.0\ folder. 

`node scripts/install.js`

ttadam, 2019-05-01:
I am having the same issue.
Thanks for the dropbox link.

rodgergr, 2019-05-02:
Same issue for me on Windows 10, but if revert to Pymakr 1.0.7, it works again. (just click on the gear symbol by the Pymakr Extension and click Install Another Version). Watch out for further automatic updates though.

Josverl, 2019-05-02:
@Ralph , 

If I turn on debug log in VSCode I see the below error 

```
Activating extension 'pycom.pymakr' failed: \\?\c:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node is not a valid Win32 application.
\\?\c:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0\node_modules\@serialport\bindings\build\Release\bindings.node.
```

If I rebuild 1.1.10 on Win10 x64 it works  , but most likely not on mac/linux...
Please feel free to download this as an interim fix 
https://github.com/Josverl/pymakr-vsc/releases/download/1.1.10_windows/pymakr-1.1.0-win.vsix

instructions for installing 
Uninstall pymakr, then use 'install from vsix' and select the file

Josverl, 2019-05-02:
@RalphHogenbirk , if I run `node scripts/install.js` in 
then I see the rebuild fails : 
```
PS C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0> node scripts/install.js                                      Copy bindings file
Installing electron rebuild
Copy completed
Rebuilding for electron 3.1.6...
{ Error: Command failed: .\node_modules\.bin\electron-rebuild -f -w serialport -v 3.1.6
- Searching dependency tree
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
Could not locate node-gyp

Error: Could not locate node-gyp
    at Rebuilder.<anonymous> (C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0\node_modules\electron-rebuild\lib\src\rebuild.js:172:23)
    at Generator.next (<anonymous>)
    at fulfilled (C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0\node_modules\electron-rebuild\lib\src\rebuild.js:4:58)

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
  killed: false,
  code: 4294967295,
  signal: null,
  cmd:
   '.\\node_modules\\.bin\\electron-rebuild -f -w serialport -v 3.1.6' }
done
PS C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.0>                                                              
```

and node-gyp is developed as cross plaform , but somhow forgot about adressing windows ....

apparently getting it to work outside of the linux code base requires luck , magic or unicorns 
https://spin.atomicobject.com/2019/03/27/node-gyp-windows/

But when it works is apparently great ( i have tried many times and not got it to work myself)

RalphHogenbirk, 2019-05-04:
@Josverl superb detective work there. In conclusion, you are right. We can't rely on node-gyp. But my assumption that trying to run electron-rebuild anyway could not hurt anything, was apparently wrong:

Precompiles have already been included (/precompiles/serialport-<os>) and the install.js copies the bindings.node file from there to the serialport folder in the node_modules. 
In install.js where it say 'Copy completed', that's where it should have been done. After that, it tries the 'electron-rebuild' command anyway, just in case vscode has been updated with a newer version of electron and pymakr is somehow not up to date anymore. But I suspect that the failing electron-rebuild command somehow reverts the bindings.node file back to a 'non-rebuild' state when attempting to re-compile again.

So, I think this could be very simple solved with this commit: 9fd6d78f93a025a2544624f00e87da63ede2f56c. But i'm not entirely sure. But at least I know how to test this now on my windows machine

Jerremy09, 2019-05-07:
Hello,

I see you comunicate this issue much here. My question is if you will create a new Release so correction will be easly updated in our VS Code?

Thank you

Jan

RalphHogenbirk, 2019-05-07:
@Jerremy09 yes I'm planning to release this new code together with one other bugfix later today or tomorrow. You'll be able to update your plugin from vscode and the issue should be fixed.

Jerremy09, 2019-05-08:
@RalphHogenbirk Thank you for confirmation. It is not crucial for me to work today. So I will wait.

RalphHogenbirk, 2019-05-08:
Update: I'm testing on another windows machine this morning and found another issue. The install script works properly when using in the dev environment. But when packaging as .vsix and installing like a normal extension, the install script fails to copy the precompiled bindings.node file. 

I've manually run the install script right after install, and got the error `EBUSY: resource busy or locked, open 'C:\Users\ralph\.vscode\extension
s\pycom.pymakr-1.1.1\node_modules\@serialport\bindings\build\Release\bindings.no
de` which I suspect is also happening during install. Perhaps because we're running it on 'install' from the package.json, it runs it post-install when vscode has already started the extension.
I'm diving in to this, but maybe @Josverl might have an idea about this too in the meantime?

RalphHogenbirk, 2019-05-08:
In any case, running the command `node scripts/install.js` from your user/.vscode/extensions/pymakr-1.1.0/ folder, should solve the issue for now @JavierNR @Jerremy09 @rodgergr @ttadam

Josverl, 2019-05-08:
Hi Ralpf , 
sorry but that workaround does not work for me as I tested before ( and verified just now ) 
```cd %userprofile%\.vscode\extensions\pycom.pymakr-1.1.0
C:\Users\<user>\.vscode\extensions\pycom.pymakr-1.1.0>node scripts/install.js
```
still results in 
``` × Rebuild Failed
An unhandled error occurred inside electron-rebuild
Could not locate node-gyp
```  
Until we have a final solution ; I would suggest other **Windows** users to use the below 1.1.0-win package that has the correct pre-compiled serial package for windows :
Please feel free to download this as an interim fix
https://github.com/Josverl/pymakr-vsc/releases/download/1.1.0_windows/pymakr-1.1.0-win.vsix
instructions for installing
Uninstall pymakr, then use 'install from vsix' and select the file.

To fully resolve this I suggest to revert to the prior method : 
- detect platform , 
- ten load the serial module from respective platform folder 
While that may not prevent future issues when vscode upgrades to a new electron version, it does solve  the current issue.

RalphHogenbirk, 2019-05-08:
You're right, install.js does throw that error, but it does actually fix Pymakr. The node-gyp error is not relevant, it copied the correct bindings file before that line in the script. If you reload vscode after running install.js, pymakr should work again (it did on my tests). (Only the version on the dev branch it skips the electron-rebuild command on windows machine)

The problem is that install.js somehow doesn't copy that file correctly when it's executed from the 'install' property in package.json when it's installing the vsix. I suspect an issue with VSCode not waiting for install.js to finish, before starting up the plugin anyway, resulting in a read conflict (EBUSY: resource busy) of the bindings.node file in the `node_modules/@serialport/bindings/Build/release` folder. Working on the issue now.

Josverl, 2019-05-08:
Aha indeed :-)

RalphHogenbirk, 2019-05-09:
I've made a workaround. If one of you would like to try it, you can download the vsix here: https://www.dropbox.com/s/nrkjv145ecj3l05/pymakr-1.1.2.vsix?dl=0
Code is on the develop branch.

Jerremy09, 2019-05-09:
Hi I tested and I can confirm that Pymakr starst within VS Code. Testes with:
Pymakr 1.1.2 (Currently I think that it should be 1.1.1 - but in your version is 1.1.2 (Change log has 1.1.1))
Pymakr 1.1.0 
- both are working with me

Josverl, 2019-05-09:
@RalphHogenbirk , Yes I can confirm that it works .

just fyi the debug logs during the install and first activation on Win10 X64 
``` 
[Extension Host] Error while loading serialport lib. Trying to fix it...
console.ts:134 [Extension Host] Serialport included without issues
console.ts:134 [Extension Host] [warning] SettingsWrapper | Error processing Config file:c:\Users\josverl\OneDrive\PowerShell\O365\Cloud_Hybrid_Search_Scripts/pymakr.conf
console.ts:134 [Extension Host] [warning] SettingsWrapper | Error opening config file 
console.ts:134 [Extension Host] [warning] SettingsWrapper | Error: ENOENT: no such file or directory, open 'c:\Users\josverl\OneDrive\PowerShell\O365\Cloud_Hybrid_Search_Scripts\pymakr.conf'
console.ts:134 [Extension Host] [warning] SettingsWrapper | Error processing Config 
```

vscode version details : 
Version: 1.33.1 (user setup)
Commit: 51b0b28134d51361cf996d2f0a1c698247aeabd8
Date: 2019-04-11T08:27:14.102Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 10.0.18362

RalphHogenbirk, 2019-05-13:
@Josverl good to hear! Yes that output looks correct. It fixes the serialport lib on first start now if needed, after that you should not see those first two lines in the output anymore. 
I've just released a new version that includes this fix. If there's any problems with it, feel free to re-open this issue!

closed 2019-05-13 by RalphHogenbirk