pycom/pymakr-vsc #43: Extension fails to load if invalid pymakr.conf is in workspace

osechet, 2019-05-04:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?

1. Open a folder in vscode
2. Create a pymakr.conf file with invalid json. In my case it was a trailing coma on the last line.
3. Reload vscode (it should reopen the same folder).

## What happens?

On restart, the pymakr extension won't load. 

## What were you expecting to happen?

The extension should load, reporting an error in the conf file and ignoring its content.

## Any logs, error output, etc?

Here are my global settings:
```
{
	"address": "/dev/tty.wchusbserial640",
	"username": "micro",
	"password": "python",
	"sync_folder": "",
	"open_on_start": true,
	"safe_boot_on_upload": false,
	"py_ignore": [
		"pymakr.conf",
		".vscode",
		".gitignore",
		".git",
		"project.pymakr",
		"venv"
	],
	"fast_upload": false,
	"sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
	"ctrl_c_on_connect": false,
	"sync_all_file_types": false,
	"auto_connect": false,
	"autoconnect_comport_manufacturers": [
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc."
	]
}
```

The developer's tools console report the error (notice the trailing comma on the last line):
```
TypeError: Cannot use 'in' operator to search for 'address' in {
    "address": "/dev/tty.wchusbserial640",
    "username": "micro",
    "password": "python",
    "sync_folder": "",
    "open_on_start": true,
    "safe_boot_on_upload": false,
    "py_ignore": [
        "pymakr.conf",
        ".vscode",
        ".gitignore",
        ".git",
        "project.pymakr",
        "venv"
    ],
    "fast_upload": false,
    "sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
}
```

## Any other comments?

## What versions of software are you using?

**Operating system:**

MacOS 10.13.6

**VSCode version:**

1.33.1

**Pymakr version:**

1.1.0

RalphHogenbirk, 2019-05-07:
I've checked this and there is indeed an error in the code that's handling format errors. I've fixed it with a few improvements (here: https://github.com/pycom/pymakr-vsc/commit/8a82fb1cf1729c33693f1b9be5e25a457d80b588) and will deploy it in a new release today or tomorrow.

RalphHogenbirk, 2019-05-16:
Fixed in the latest release

closed 2019-05-16 by RalphHogenbirk