pycom/pymakr-vsc #44: Other ESP32 boards

Jerremy09, 2019-05-08:
Hello all,

I want to ask you to take a look to video attached, where I connect ESP32 board to my PC and Flash it by micropython Firmware (not in video), but I want to upload my main.py into board but unfortunatelly with no access. VS Code fall every time. 

1) I tested almost all Firmware version 
2) VS Code updated to version: 1.33.1
3) I corrected Pymark 1.1.0 by correction set in this issue (@RalphHogenbirk solution):
    "In any case, running the command node scripts/install.js from your user/.vscode/extensions/pymakr-1.1.0/ folder, should solve the issue for now @JavierNR @Jerremy09 @rodgergr @ttadam"
4) Node.js: 10.15.1
5) Architecture: x64

but is it even possible to operate different ESP board? 
1) REPL - works
2) Run - works
3) Upload - do not work

Link: https://www.youtube.com/watch?v=gXn_wUPcgmw&feature=youtu.be

Thank you for help

Jan

RalphHogenbirk, 2019-05-09:
HI @Jerremy09, interesting case. We normally don't actively test that much with other boards than Pycom boards for compatibility, but where we can we try to keep it compatible. In your case I suspect that your board might not support the safe-boot function (ctrl-f) that Pymakr uses before upload. Try disabling 'safe boot before upload' in the settings and try it again.

Jerremy09, 2019-05-09:
Hello @RalphHogenbirk thank you for advice, I can confrim that first upload to empty ESP32 works fine, but any other uploead is inpossible, simmilar issues like before :-(. you can take a look on Youtube again (there are 2 videos - I cutted flashing firmware).

1) New Firmware - done by uPyCraft - esp32-20190508-v1.10-332-g3f54462ad
2) VS Code - 1.33.1
3) Pymakr - both 1.1.0 and 1.1.2 (same results)

Link: https://www.youtube.com/watch?v=dDmlwMOqL9o&feature=youtu.be

Josverl, 2019-05-09:
Hi @Jerremy09 ,

I only work COM12I work with a decent range of different ESP32 boards , and with the recent versions of pymakr I am able to connect to all hardware I have , and quite some firmwares as well.
( Lobo , M5Stack , micropython 8266 , micropython esp32 all seem to work.

could I ask you  :  
* a link to the FW you flashed.

I know from having tried my hand at uPycraft that they added a their own upload/download .py modules to the board to support their onw upload download.  maybe that is the problem.

can you also give a miminal step by step repro of the problem. I appreciate the  YT, but its pretty hard to follow to be honest.

ie something like : 

1. flash board wit uPycraft 
2. start code ./repro 
2. create something.py and save it 
3. Press connect connect  
4. upload 
....

does that make sense ?

Jerremy09, 2019-05-10:
Hello @Josverl thank you to step into conversation. Okay I will try to provide maximum data :-)

1) uPycraft - I use it only for flash Firmware to ESP board (I select downloaded one, because I can flash with newer)
2) Firmware used - I tested all of these and the result is same
   A) esp32-20190508-v1.10-332-g3f54462ad
   B) esp32-20190125-v1.10
   C) esp32-20180511-v1.9.4
    - but all of them were downloaded from "http://micropython.org/download#esp32"
3) VS Code - 1.33.1
4) Pymakr - both 1.1.0 and 1.1.2
5) NodeJs - 10.15.1
6) Woindows - 10 (64)

Test steps:
1) Flash firmware from PC by uPycraft:
   A) Erase flash = Yes
   B) Burn Address = 0x1000
   C) select one from mentioned above

2) After succesfull burning I turn off upycraft
3) Open VS Code - thanks to correction the Pymakr is loaded correctly and I have these Global settings (no project settings)
{
	"address": "COM8",
	"username": "",
	"password": "",
	"sync_folder": "",
	"open_on_start": true,
	"safe_boot_on_upload": false,
	"fast_upload": true,
	"sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
	"ctrl_c_on_connect": true,
	"sync_all_file_types": false,
	"auto_connect": false,
	"py_ignore": [],
	"autoconnect_comport_manufacturers": [
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc."
	]
}
         Safe_boot_on_upload is turned off because it helped me to uload for first time (thanks @RalphHogenbirk )

4) Project in VS Code contain: Boot.py and Main.py (there is code to blink onboard diod only)
   My code in main.py:
   --------------------------------------
   import time
   from machine import Pin
   LED = Pin(1,Pin.OUT)
   citac = 0

   while citac<20:
     LED.value(1)
     time.sleep(0.2)
     LED.value(0)
     time.sleep(0.2)
     citac = citac + 1
--------------------------------------
5) At this point I can hit "run" function several times and Boar will do the stuf written in "main.py"
  - there is only one need to restart board after every run an press "Run" button again
6) if I upload code it upload it to board correctly , but only when the board is flashed and without any project and it gives me this:
--------------------------------------
Uploading to /...
Reading file status
Failed to read project status, uploading all files
[1/2] Writing file boot.py (0kb)
[2/2] Writing file main.py (0kb)
Upload done, resetting board...
OKets Jun  8 2016 00:22:57
--------------------------------------

7) Then I correct code in VS Code (change time.sleep() function) - but this cannot be uploaded the code to board, because I have this (there are no lines with firmware...):
   --------------------------------------
   ...
   I (578) cpu_start: Pro cpu start user code
   I (149) cpu_start: Starting scheduler on PRO CPU.
   I (0) cpu_start: Starting scheduler on APP CPU.
   E (10) spiram: SPI RAM not initialized
   --------------------------------------

When I restart board with only flashed firmware I have this (sory I canot upload picture here):
   --------------------------------------
   I (578) cpu_start: Pro cpu start user code
   I (149) cpu_start: Starting scheduler on PRO CPU.
   I (0) cpu_start: Starting scheduler on APP CPU.
   E (10) spiram: SPI RAM not initialized
   MicroPython v1.10-336-gc0a1de3c2 on 2019-05-10; ESP32 module with ESP32
   Type "help()" for more information.
   >>>
   --------------------------------------

Thank you

Jan

Josverl, 2019-11-11:
I assume this issue can be closed now ?

closed 2019-11-11 by pedroknup