pycom/pymakr-vsc #45: Error "command 'pymakr.globalSettings' not found" in VSC 1.33.1 + latest plugin version

choeffer, 2019-05-09:
## What happens?

After installing all above I tried to use "Pymakr > Global settings" to setup my connection to the LoPy4 but I just get the error message "command 'pymakr.globalSettings' not found".
I also tried to use other pymakr commands from the Command Palette and with all commands I get this kind of error. Also in the left down corner there is no pymakr related gui elements which should be there.

## What were you expecting to happen?

Getting the gui elements and that the commands are working.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

Not yet but I am willing to help and to provide logs if needed.

## Any other comments?

Installed Software:

VSC (output of "Help" -> "About"):
Version: 1.33.1
Commit: 51b0b28134d51361cf996d2f0a1c698247aeabd8
Date: 2019-04-11T08:20:22.771Z
Electron: 3.1.6
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Linux x64 4.19.0-4-amd64

node.js LTS version:
https://packages.debian.org/de/buster/nodejs

npm (just in case):
https://packages.debian.org/de/buster/npm

## What versions of software are you using?

**Operating system: Debian Buster all updates installed**

**VSCode version:1.33.1**

**Pymakr version: 1.1.0**

RalphHogenbirk, 2019-05-09:
I think this is the same as what's happening in this issue: https://github.com/pycom/pymakr-vsc/issues/42. We're working on a fix!

RalphHogenbirk, 2019-05-09:
Just to confirm, if you open the developer tools and check the console, do you see an error that the bindings.node file is not valid?

choeffer, 2019-05-09:
As you suggested in #11 , https://github.com/pycom/pymakr-vsc/issues/42#issuecomment-490413721 results in:

```
pink@debian-t470:~/.vscode/extensions/pycom.pymakr-1.1.0$ node scripts/install.js
Copy bindings file
Installing electron rebuild
Copy completed
Rebuilding for electron 3.1.6...
{ Error: Command failed: $(npm bin)/electron-rebuild -f -w serialport -v 3.1.6
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
- Searching dependency tree
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
Could not locate node-gyp

Error: Could not locate node-gyp
    at Rebuilder.<anonymous> (/home/pink/.vscode/extensions/pycom.pymakr-1.1.0/node_modules/electron-rebuild/lib/src/rebuild.js:172:23)
    at Generator.next (<anonymous>)
    at fulfilled (/home/pink/.vscode/extensions/pycom.pymakr-1.1.0/node_modules/electron-rebuild/lib/src/rebuild.js:4:58)

    at ChildProcess.exithandler (child_process.js:294:12)
    at ChildProcess.emit (events.js:189:13)
    at maybeClose (internal/child_process.js:970:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
  killed: false,
  code: 255,
  signal: null,
  cmd: '$(npm bin)/electron-rebuild -f -w serialport -v 3.1.6' }
done
```

But the dependency for the npm version in Debian Buster is "dep: nodejs (>= 6.11~)", so this should not be an issue.

Any ideas?

RalphHogenbirk, 2019-05-09:
@choeffer Correct, but very likely your issue is solved anyway. The lines `Copy bindings file` and `Copy completed` are the ones that matter. Try to reload vscode and see if Pymakr works again.

choeffer, 2019-05-09:
You are right. I get the following terminal output.

```
Welcome to the Pymakr plugin! Use the buttons on the left bottom to access all features and commands.
This is how you get started:
 1: Open 'Global Settings' (we went ahead and did that for you)
 2: Connect a pycom board to your USB and the terminal will auto-connect to it (you can skip step 3 and 4 now)
 3: If you want to connect over WiFi, disable auto_connect and fill in the correct IP or serial port of your Pycom board in 'address'
     (When using serial, you can also use the 'List serial ports' to find the correct serial port)
 4: Connect using the 'Connect' command or the 'Pycom Console' button
 5: Open a micropython project with main.py and boot.py files
 6: Start running files and uploading your code 

 Use the 'Help' command for more info about all the options 

AutoConnect enabled, ignoring 'address' setting (see Global Settings)
Searching for PyCom boards on serial...
No PyCom boards found on USB
```
Could this error about rebuilding for electron from above still be an issue when installing directly from the marketplace? How could I narrow it down?

RalphHogenbirk, 2019-05-09:
That's exactly the issue. I've made a workaround now (you can test it if you like, check my answer here: https://github.com/pycom/pymakr-vsc/issues/42#issuecomment-490939536). Will release that if it works for a few testers.

RalphHogenbirk, 2019-05-16:
This issue was solved in the latest release

closed 2019-05-16 by RalphHogenbirk