pycom/pymakr-vsc #46: Error "Running the contributed command: 'pymakr.upload' failed."

choeffer, 2019-05-09:
## What are the steps to reproduce this issue?
1. Open VSC
2. Connect LoPy4 + Pytrack board vis USB
3. Try to upload or download files to/from the LoPy4

## What happens?

Executing the code to let the led blink works fine ("Pymakr > Run Current File"). But I am not able to upload or download files via USB. Connecting via Wifi and then FTP works fine.

## What were you expecting to happen?

Uploading and downloading from the LoPy4 should work.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

See attached screenshot

## Any other comments?

## What versions of software are you using?

**Operating system: Debian Buster all updates installed**

**VSCode version: 1.33.1**

**Pymakr version: 1.1.0**

choeffer, 2019-05-09:
[image]

RalphHogenbirk, 2019-05-09:
Thanks for the report, there's definitely something wrong. One thing to check before we dive in to solve this: did you press 'upload' very fast after load, perhaps before the plugin was fully loaded up? I see some terminal errors as well, which usually means the plugin is still starting up.

choeffer, 2019-05-09:
I tried it again in a fresh VSC session and waited a bit before trying to download files, see attached screenshots. Are they usefull or is more information needed?
[image]
[image]

RalphHogenbirk, 2019-05-09:
Thanks, that excludes that scenario. Could you perhaps post the contents of your global settings file? I'm trying to replicate the issue with the same setup, but i'm unsuccessful so far.

choeffer, 2019-05-09:
Here is my pymakr.json. Hope this helps to narrow  it down. If you need more information, please let me know.

```
{
	"address": "192.168.4.1",
	"username": "micro",
	"password": "python",
	"sync_folder": "",
	"open_on_start": true,
	"safe_boot_on_upload": false,
	"py_ignore": [
		"pymakr.conf",
		".vscode",
		".gitignore",
		".git",
		"project.pymakr"
	],
	"fast_upload": false,
	"sync_file_types": "py,txt,log,json,xml,html,js,css,mpy",
	"ctrl_c_on_connect": false,
	"sync_all_file_types": false,
	"auto_connect": true,
	"autoconnect_comport_manufacturers": [
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc."
	]
}
```

RalphHogenbirk, 2019-05-09:
Thanks, there seems nothing wrong with that. I have pretty much the same setup, but I can't reproduce anything. Only thing is python 3.7 vs python 2.7, but that should not matter.
Can you tell me which board you are using and which firmware it is running?

choeffer, 2019-05-09:
Python 3 vs Python 2 can have an influence. Some functions may behave different or are not existing or deprecated. Could you please also check it with Python 3?
And in general, especially because MicroPython is based on Python 3, Python 2 is nearly EOL https://pythonclock.org/ and Python 3 is already over 10 years on the market, I suggest to switch to Python 3. Which parts are relying on Python 2?
The firmware on the Pytrack extension board is pytrack_0.0.8.dfu and the one on the LoPy4 is 1.18.2.r6.

RalphHogenbirk, 2019-05-09:
I will, but actually I was referring to the python version on my windows machine. Pymakr is not running any python code on the machine (in this case), just on the micropython board. So it shouldn't matter :)

About the LoPy4 firmware: I think there's only v1.18.2.r2 and v1.18.1.r6. If you're running the latter, might be better to update. (v1.20.0.rc8 might also be a nice option since it supports compression and you can enable 'fast upload' in pymakr)

choeffer, 2019-05-09:
Okay, then it should not have an effect :) Was just a bit confused that people still use Python 2 nowadays.

I tried the same like above with a Python 2.7 environment in VSC, but the outcome was the same (was testing in parallel, but after reading what you said it should also not work).

The terminal prints out "Pycom MicroPython 1.18.2.r6". I used "Pycom Upgrade" installed via .deb package and this installed this firmware version yesterday if selecting the stable channel.

[image]

RalphHogenbirk, 2019-05-09:
An understandable confusion :)

My mistake, I guess 1.18.2.r6 was not tagged in the firmware repo, but then your firmware is indeed the latest stable and should also not be causing any issues.
I have a day off work tomorrow but I'll try to do some more testing, otherwise continue on monday with this. I'm planning to do a new release asap, and if this is some edge-case bug then I want to include it as fast as possible.

Things you can try in the meantime to get you up and running:
- Try Pymakr for Atom, see if that has the same issues. That's useful info I can use to narrow down the search.
- Downgrade to 1.0.7 (https://www.dropbox.com/s/41td19stzgk8fn2/pymakr-1.0.7.vsix?dl=0)
- Try a different project folder and clear the board (os.fsformat('/flash')

I'll let you know as soon as I find any possible cause.

choeffer, 2019-05-09:
Thanks for all your efforts and the fast replys :)

I will try to go through your suggestions. I have the feeling that the project folder might cause an issue because it is not an empty one or does not have the exact structure as mentioned in the wiki https://docs.pycom.io/gettingstarted/programming/first-project.html . My workspace includes also a latex folder etc., as you can see on the screenshots. Maybe the best will be to start with an empty worspace to narrow it down.

I will let you know if I find a reason or if one of your suggestions helped.

choeffer, 2019-05-10:
The cause was the workspace structure. I created an empty folder called "pycom", then in a terminal

```
cd pycom
code .
```

to start VSC in that folder and after connecting the Pytrack + LoPy4 via USB I was able to download the files.

[image]

choeffer, 2019-05-10:
I was able to break it again. Seems like a Python3 venv in the same folder breaks the command.

[image]

VSC is able to detect a venv https://code.visualstudio.com/docs/python/environments . Maybe it gets selected/detected by the plugin via VSC and then fails because it is not activated.

[image]

Hope you can figure it out/solve it, because venvs should not break the plugin.

choeffer, 2019-05-10:
Update:

Not the loaded venv is the problem. I placed the venv folder outside the project folder and linked to it via the workspace settings so that the venv is activated in VSC, see screenshot. Even then the dowload is working. So it seems that placing the venv folder inside the project folder is causing the issue, not activating the venv in VSC.

[image]

RalphHogenbirk, 2019-05-13:
@choeffer Thanks for testing that, glad you figured it out! I've reproduced it by adding the venv folder and found an issue in my code with ready files without an extension (which are present in the venv/bin folder). So i've fixed that and have also added 'env' and 'venv' to the default ignore list. So should definitely be solved now. I'll release it in the new version later today.

choeffer, 2019-05-13:
@RalphHogenbirk Thanks for sorting it out and fixing it. But be aware, that the folder which contains a venv can have any freely chosen name, not just _venv_ . But there should exist a reliable method to detect a venv because vsc can also detect them if they are in the same folder, independently of the name.

choeffer, 2019-05-13:
@RalphHogenbirk And there can be different kinds of them, see https://code.visualstudio.com/docs/python/environments#_where-the-extension-looks-for-environments . I use the "standard" Python3 once, but there are also other options available to create them, so maybe you can find the code in the Python extension which does the recognition and you can reuse it.

RalphHogenbirk, 2019-05-13:
Yes that's true! But it's ok, if it's named differently, the plugin won't crash anymore. That issue was fixed. Adding it to the ignore file is just an extra convenience :)
Looks like I could read out the python workspace settings and read out the 'venv' folder to ignore it automatically. Might do that for the next release!

RalphHogenbirk, 2019-05-13:
Meanwhile, 1.1.2 was released, solving this issue :) You'll probably have to add venv to your py_ignore setting manually though, if you don't want any files in that folder to upload to your board.

closed 2019-05-13 by RalphHogenbirk

choeffer, 2019-05-14:
Thanks again :) Update was automatically installed today. I will report back if there still exists problems regarding this issue. And thanks for the hint with the py_ignore setting.
Maybe you should add that never a venv folder is synced to a pycom board, because the venv is just meant to be used locally on the computer and never on the pycom device itself. This would avoid users run into the issue that a venv folder is synced and then the internal memory gets filled up by it. The size of my venv is e.g. 6 MB.