pycom/pymakr-vsc #47: Too many Pycom Console terminals are created which can not be killed

choeffer, 2019-05-13:
## What are the steps to reproduce this issue?

1. Open VSC
2. Connect LoPy4 + Pytrack board vis USB
3. Run different commands from the pymakr plugin

## What happens?

More than one _Pycom Console_  is created but they can not be closed. If trying to close them via the _Kill Terminal_ button in VSC in the right down corner, they get recreated even if they do not have any use.

## What were you expecting to happen?

Just one _Pycom Terminal_ should get created.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

[image]
[image]
[image]
[image]

## Any other comments?

## What versions of software are you using?

**Operating system: Debian Buster all updates installed**

**VSCode version: 1.33.1**

**Pymakr version: 1.1.0**

osechet, 2019-05-15:
If it can help, I found an easy way to reproduce:
1. Open vscode and wait for the first Pycom console to open
2. Open a "standard" vscode terminal by pressing the '+' button
3. Kill the "standard" terminal
=> a new Pycom console opens that cannot be killed.

RalphHogenbirk, 2019-05-16:
Thanks for this report, and @osechet for the hint that made me find the issue. Turns out the 'onClose' callback from the vscode.terminal API was not checking if the closed console was actually the Pycom console or another one. I've pushed a fix to the develop branch now. Now it just restarts the Pycom Console when it's closed, not when any other terminals are closed. It'll be included in the next release

closed 2019-05-16 by RalphHogenbirk

Josverl, 2019-05-20:
thanks for finding (and fixing) this , I ran into it all the time but never found the repro steps ...

romzes13, 2020-06-10:
The issue is still there. Somehow Pycom Console just wont die. Tried killing it in VS code and restarting IDE without success.
Every time I start/restart my IDE the console opens up and tries to connect to USB
```
AutoConnect enabled, ignoring 'address' setting (see Global Settings)
Searching for PyCom boards on serial...
No PyCom boards found on USB``` 

I should probably add a new issue.

Josverl, 2020-06-10:
@romzes13 if pymakr is enabled , it will always create a terminal.
as i use vscode all day long, and pymakr only for personal work sometimes it does get in the way. ( as is the case with some of the other 60 extensions i use. 
i make use vscode`s configuration options to deal with this 

- pymakr is installed , but disabled globally (1 tmie action) 
- for each micropython prject I enable it for that folder / workspace ( 1 time action per folder / project) 
[image]

and it works where I want it , and does not get in the way in any other project