pycom/pymakr-vsc #48: Intellisense/linting for pycom libraries

choeffer, 2019-05-15:
# Feature Request 🚀

## Describe the solution you'd like

https://code.visualstudio.com/docs/editor/intellisense / https://code.visualstudio.com/docs/python/linting also for modules/classes from the MicroPython and Pycom libraries. Without code completion / linting everything has to be looked up in the docs and even simple mistakes are not highlighted by vsc as they normally would be via pylint.

## Teachability, Documentation, Adoption, Migration Strategy

Maybe this can be done with a venv where the pycom modules/classes are imported/installed so that they are found by the vsc Python extension. I also found https://github.com/Josverl/micropython-stubber which can be used/extended for this.

Josverl, 2019-05-20:
:-) Yes this is indeed something that I have been working on.
not sure if it should be a seperate add-in , part of pymakr or part of the python add-in 

for now looking for a way to have an add-in automatically configure options for other add-ins ; 
ie for addin foo-bar to set options for pylint (.pylintrc) and the python add-in (in settings.json)

RalphHogenbirk, 2019-05-25:
Would be awesome, had more people asking for this. Let me know if you have any progress on this @Josverl. A separate add-in would be nice. Possibly with Pymakr detecting if it's installed and advising the user to do so? Making it part of Pymakr would also work for me, but separating it would make more sense.

BradenM, 2019-05-28:
@RalphHogenbirk @Josverl I've started something at [BradenM/micropy-cli](https://github.com/BradenM/micropy-cli) that may be useful towards this. It makes use of Josverl's createstubs.py to automate stub creation (from any plugged in pyboard-like device) and manage any other stubs you may use. Then when you add stubs, running `micropy init <project>` allows you to select the stubs you wish to use and creates a project folder with them setup via .vscode and .pylintrc as described in [micropython-stubber](https://github.com/Josverl/micropython-stubber). It's very early on as of right now, but wanted to see what you guys think before I continue it's development.

Josverl, 2019-05-28:
Hi Braden, 
That looks cool. I added Frozen modules to improve Intellisense, and there i found that its even more important to have the exact correct order and Casing in the config files. 
so any help our whetware can get will only improve matters.

@Ralph a seperate add-in would make sense, but I wonder if it is possible to call some pymakr functions from my add-in ?  ie to ask pymakr to run a .py file on the MCU , and to download generated the stubs to a folder ?

But I think I'll start with the configuration and download of the stubs parts first ;-)

BradenM, 2019-05-29:
@Josverl,
Nice, I will definitely take a look at the frozen modules. Thanks.

Also, perhaps I was a bit unclear when I said "automate stub creation" earlier. Except for the calling Pymakr part, `micropy stubs create <port>` actually does exactly what you just said. (as can be seen [here](https://github.com/BradenM/micropy-cli/blob/7fad6ab7579052abe618e0718d852b78268e655e/micropy/micropy.py#L61)). Its currently a bit rough and needs some work, but running the command does the following:
* Downloads createstubs.py from [your repo](https://github.com/Josverl/micropython-stubber/blob/master/createstubs.py)
* Uses rshell to connect and upload createstubs.py
* Uses rshell to execute createstubs.py
* Recursively downloads the generated `stubs` folder
* Adds the new stubs to `$HOME/.micropy/stubs` folder, which can then be listed via `micropy stubs list`

Is this close to what you had in mind?

askpatrickw, 2020-03-20:
Until this gets sorted out... I made the GPY version
https://github.com/askpatrickw/pycom-stubs/

Josverl, 2022-12-04:
https://github.com/Josverl/micropython-stubs#micropython-stubs