pycom/pymakr-vsc #53: There was an error with your serialport module

beaniebag, 2019-07-04:
Each time I run Visual Studio Code I get the following error message now:

`There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)`

Vs Code details are:
Version: 1.36.0 (user setup)
Commit: 0f3794b38477eea13fb47fbe15a42798e6129338
Date: 2019-07-03T13:25:46.372Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.17134
Pymakr: 1.1.2

I've tried uninstalling and reinstalling previous versions but that just removes the error and pymakr doesn't load.

10caret100, 2019-07-04:
Same issue here. I spent the better part of 4+ hours troubleshooting a tool, which is supposed to be way out of Beta already.

It actually (kind of) worked on very first install (PyMakr 1.1.2), but was very fragile and would constantly lose connection to the PyCom boards (PySense w/ FyPi) and I'd have to restart VSC, and re-connect, and do a little magic dance, etc., etc., to make it work again. 

Please stabilize this tool, as it is representative of PyCom's ecosystem and it being in such a fragile state is not good for the overall perception of the company.

The PyCom modules are very promising, indeed, but without a stable dev env, I don't think we'll consider using the modules for our current project (which, if PoC work proves them to be suitable, we'll likely order 1000s of units in the future). Stable dev tools (that, at the very least, install without errors) is a strict passing criteria for making this decision, though.

Tech info:

Version: 1.36.0 (user setup)
Commit: 0f3794b38477eea13fb47fbe15a42798e6129338
Date: 2019-07-03T13:25:46.372Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.17763

Josverl, 2019-07-04:
Apparently this is caused by VSCode switching to a newer version of Electron, which requires a different binding for the serialport module in order to work.
that the ability to connect, and to connect reliably is something that should be addressed. In fact there has been quite some effort spend on this in the past to address similar issues in previous version jumps in VScode / electron, but It seems that the additional logic (ref: #42)  that was included to handle these , and select the correct bindings did not do its job in the current instance. 

Making cross-platform serial port work for any future version of Electron is notoriously challenging.

Perhaps an automated integration test running against VSCode Insider would at least provide advance warning for the DevTeam to prepare, and include the new bindings in advance.

10caret100, 2019-07-04:
I understand the conundrum, but, ideally, PyCom should come up with a solid workaround until this issue is addressed adequately in a future release. Free free to let us know if there's anything the community can do to help? Which parts of the code base are the offending ones?

beaniebag, 2019-07-04:
Currently the work around for me is to use another IDE.  I've been using atom in the mean time till a fix comes out for vs code version.

Josverl, 2019-07-04:
Re ert to a previous version of vscode is an option, but not simple.
I'll see if I can build a private version for Windows for the interim. If that works I'll post here.

GC-RnD, 2019-07-05:
simply put... when can we expect a fix

Josverl, 2019-07-05:
well , i'm only a member of the community, so I have no SLA, 
If you are a paying customer than I assume you can call pycom support to log a ticket.

meanwhile I've been doing some diffing into the wonders of electron and the integration of modules in a multiplatform environment.

if you open a command window and paste the below commands,
that should download & install a better version of the serialport\binding that offers the integration into electron , whish is version specific. 
4.2.5 is the Electron version that you can also see in the **VSCode > Help > About** 

``` 
cd %userprofile%\.vscode\extensions\pycom.pymakr-1.1.2
cd node_modules\@serialport\bindings
.\node_modules\.bin\prebuild-install.cmd --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose --force
```
the output should look something like this 
```
prebuild-install info begin Prebuild-install version 5.3.0
prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\a3ab8a-bindings-v2.0.8-electron-v69-win32-x64.tar.gz
prebuild-install info found cached prebuild
prebuild-install info unpacking @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\a3ab8a-bindings-v2.0.8-electron-v69-win32-x64.tar.gz
prebuild-install info unpack resolved to C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.2\node_modules\@serialport\bindings\build\Release\bindings.node
prebuild-install info install Successfully installed prebuilt binary!
```
restart vscode, and pymakr should start working again

JoeCharlieH, 2019-07-05:
> well , i'm only a member of the community, so I have no SLA,
> If you are a paying customer than I assume you can call pycom support to log a ticket.
>
> meanwhile I've been doing some diffing into the [...]

I couldn't execute the last command in terminal:
`.\node_modules\.bin\prebuild-install.cmd --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose --force`

JoeCharlieH, 2019-07-05:
Finally managed to get it working.
Thanks for the temporal solution @Josverl.

Apparently, you need this dependencies to solve the problem: **prebuild**, **prebuild-install**
```
npm install -g prebuild
npm install -g prebuild-install
```

Next, you navigate to the bindings folder of the pymakr module:
`cd %userprofile%\.vscode\extensions\pycom.pymakr-1.1.2\node_modules\@serialport\bindings `

Finally, just past this command on the terminal:
`prebuild-install --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose`

Josverl, 2019-07-05:
Thanks Jose, 
prebuild-install can be in many different locations , it was included on my machine so i used what i though was the minimal solution  , but now i think that was some pollution from earlier attempts on my machine :-( 
I would assume you only need `npm install -g prebuild-install`  but the prebuild module will do no harm

Josverl, 2019-07-05:
@RalphHogenbirk , I now see 3 possible patterns for a more robust solution: 
1) run the `prebuild-install --runtime electron --target X.Y.Z --tag-prefix @serialport/bindings@ ` as part of the installation.
Only when VSCode skips to a new electron version users will need to re-install 
2) run `prebuild-install -- ...` each time the extention starts.
on upgrade users still get one-error / warning, or the code issues a `>reload window` command to
3) include run `prebuild-install -- ...` as part of the dev / build process : download and include multiple prebuilds ( win / linux / mac) and for multiple electron versions. and (dynamically) load the correct modules for the platform and electron version 
this will make the package slightly larger, but I think it has the best user experience

ferrarimarco, 2019-07-08:
Hi! I'm having the very same issue on VS Code 1.36.0 on MacOS. Any ETA for a stable fix? Thanks!

RalphHogenbirk, 2019-07-08:
Only seeing this now, sorry guys! VScode drastically increased the electron version in the last release (3.1 to 4.2) so the serialport build is failing. I'm rebuilding it now, will try to get a quick fix out today or tomorrow.

apeiris, 2019-07-08:
> well , i'm only a member of the community, so I have no SLA,
> If you are a paying customer than I assume you can call pycom support to log a ticket.
> 
> meanwhile I've been doing some diffing into the wonders of electron and the integration of modules in a multiplatform environment.
> 
> if you open a command window and paste the below commands,
> that should download & install a better version of the serialport\binding that offers the integration into electron , whish is version specific.
> 4.2.5 is the Electron version that you can also see in the **VSCode > Help > About**
> 
> **_updated ( thanks Jose)_**
> 
> ```
> npm install -g prebuild-install
> cd %userprofile%\.vscode\extensions\pycom.pymakr-1.1.2
> cd node_modules\@serialport\bindings
> prebuild-install --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose --force
> ```
> 
> the output should look something like this
> 
> ```
> prebuild-install info begin Prebuild-install version 5.3.0
> prebuild-install info looking for cached prebuild @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\a3ab8a-bindings-v2.0.8-electron-v69-win32-x64.tar.gz
> prebuild-install info found cached prebuild
> prebuild-install info unpacking @ C:\Users\josverl\AppData\Roaming\npm-cache\_prebuilds\a3ab8a-bindings-v2.0.8-electron-v69-win32-x64.tar.gz
> prebuild-install info unpack resolved to C:\Users\josverl\.vscode\extensions\pycom.pymakr-1.1.2\node_modules\@serialport\bindings\build\Release\bindings.node
> prebuild-install info install Successfully installed prebuilt binary!
> ```
> 
> restart vscode, and pymakr should start working again

Josverl, Managed to get it back working with the instruction.. Thanks

GC-RnD, 2019-07-08:
apeiris,

Thank you for the update my VSC Pymakr  is now function again.

Coollision, 2019-07-12:
hey guys i'm having the same issue on mac, but wasn't able to get it working within a reasonable time, so I switched to atom 😢 is there an eta on when this will be fixed for mac?

RalphHogenbirk, 2019-07-12:
A fix for windows64 and mac has been released now, together with a few small bugfixes that were pending. Hope that helps you guys for now, working on a faster solution for next time.

Coollision, 2019-07-12:
Indeed it works now, thanks for the quick action!

thangtom3, 2019-07-12:
I have same problem in linux. I've tried reinstalling newest versions but pymakr doesn't work. How i can fix this ? help me please
Version: 1.36.1
Commit: 2213894ea0415ee8c85c5eea0d0ff81ecc191529
Date: 2019-07-08T22:55:08.091Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Linux x64 4.15.0-54-generic

10caret100, 2019-07-12:
I must be doing something fundamentally wrong:

> \\?\C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\@serialport\bindings\build\Release\bindings.node
> 	at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:160:31)
> 	at Object.Module._extensions..node (internal/modules/cjs/loader.js:722:18)
> 	at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:160:31)
> 	at Module.load (internal/modules/cjs/loader.js:602:32)
> 	at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
> 	at Function.Module._load (internal/modules/cjs/loader.js:533:3)
> 	at Function.define._installInterceptor.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:491:111)
> 	at Function.t._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:760:742)
> 	at Function.t.getExtensionPathIndex.then.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:685:176)
> 	at Module.require (internal/modules/cjs/loader.js:640:17)
> 	at n (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:488)
> 	at bindings (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\bindings\bindings.js:81:44)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\@serialport\bindings\lib\win32.js:1:191)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\@serialport\bindings\lib\win32.js:111:3)
> 	at Module.t._initNodeRequire.o._compile (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:713)
> 	at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
> 	at Module.load (internal/modules/cjs/loader.js:602:32)
> 	at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
> 	at Function.Module._load (internal/modules/cjs/loader.js:533:3)
> 	at Function.define._installInterceptor.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:491:111)
> 	at Function.t._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:760:742)
> 	at Function.t.getExtensionPathIndex.then.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:685:176)
> 	at Module.require (internal/modules/cjs/loader.js:640:17)
> 	at n (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:488)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\@serialport\bindings\lib\index.js:6:22)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\@serialport\bindings\lib\index.js:17:3)
> 	at Module.t._initNodeRequire.o._compile (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:713)
> 	at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
> 	at Module.load (internal/modules/cjs/loader.js:602:32)
> 	at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
> 	at Function.Module._load (internal/modules/cjs/loader.js:533:3)
> 	at Function.define._installInterceptor.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:491:111)
> 	at Function.t._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:760:742)
> 	at Function.t.getExtensionPathIndex.then.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:685:176)
> 	at Module.require (internal/modules/cjs/loader.js:640:17)
> 	at n (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:488)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\serialport\lib\index.js:2:17)
> 	at Object.<anonymous> (C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\node_modules\serialport\lib\index.js:17:3)
> 	at Module.t._initNodeRequire.o._compile (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:713)
> 	at Object.Module._extensions..js (internal/modules/cjs/loader.js:704:10)
> 	at Module.load (internal/modules/cjs/loader.js:602:32)
> 	at tryModuleLoad (internal/modules/cjs/loader.js:541:12)
> 	at Function.Module._load (internal/modules/cjs/loader.js:533:3)
> 	at Function.define._installInterceptor.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:491:111)
> 	at Function.t._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:760:742)
> 	at Function.t.getExtensionPathIndex.then.i._load (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:685:176)
> 	at Module.require (internal/modules/cjs/loader.js:640:17)
[... cut ...]
> 	at n (c:\Users\user\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:15:488)
> 	at C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\pymakr.js:15:27
> 	at C:\Users\user\.vscode\extensions\pycom.pymakr-1.1.3\pymakr.js:158:21
> 	at ChildProcess.exithandler (child_process.js:306:5)
> 	at ChildProcess.emit (events.js:182:13)
> 	at maybeClose (internal/child_process.js:962:16)
> 	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
> 	at Socket.emit (events.js:182:13)
> 	at Pipe.Socket._destroy._handle.close (net.js:606:12)

RalphHogenbirk, 2019-07-12:
@10caret100 that's a strange error, are you on windows 64 or 32? Try completely removing any pymakr folders from C:\Users\user\extensions\, restart vscode and install pymakr again. that forces a completely fresh install.

@thangtom3 both my linux virtual machines fail to work right now, so I couldn't build the precompiled serialport lib for linux yet. Hold on for a little longer while I try to get it working! 

In the meantime, if you have time, try some of the instructions posted above to try and rebuild it manually. You can also try to run scripts/rebuild.js, but that'll likely fail on your machine.

10caret100, 2019-07-12:
@RalphHogenbirk I'm using **Win10 x64 Home**

If you mean C:\Users\user\.vscode\extensions - already tried that - no go :( 
Should I try using a 32-bit version of NodeJS, and if yes - which version, should I use?

Thanks

RalphHogenbirk, 2019-07-12:
@10caret100 no, the 64 -bit is the one that should work properly. I can't find the actual error in that massive stacktrace you posted. Did you mis a line when copying? Can you give me a screenshot of pymakr when this happens? Also, which versions of Pymakr, Vscode and Node are you using?

Josverl, 2019-07-12:
@ralphHogenbirk, if you follow my recipe above you can download any binding for Linux on your mac

BradenM, 2019-07-26:
Was also facing this issue in arch linux. Followed @Josverl's build instructions (adapted for linux) recompiled `bindings.node`. Will open a PR with the new binary.

wadebee, 2019-07-26:
Having this same problem with PyMakr 1.1.3 against VS Code 1.36 .1 running under Ubuntu Linux. Would you please post the steps needed for this fix?

BradenM, 2019-07-26:
@wadebee, 

I updated the PR I made above (#57) with the exact steps I used to get it working. They are nearly identical to what Josverl stated above.

wadebee, 2019-07-27:
Thanks Branden. I followed your steps but used the npm install option list by Josverl. All steps worked fine for me and now when I open VSCode I no longer get the serial port error. 

However now I am getting an error from VS Code about NodeJs not being found on my machine. It is installed (latest 12.7 and also LTS 10.16) so I don't know why its complaining. I installed Node via nvm script rather that apt-get. Does that matter?

ftdomingo, 2019-08-05:
This is magic. Thank you for the work around!

Josverl, 2019-08-05:
I have been working on a PoC to prevent this problem in the future, and from my initial that seems to be working fine.
essentially this will download and include the prebuilds, for multiple platforms, and electron versions into the extention and 'senses' future electron versions used vscode by looking at github.
there is still a lot of documentation, testing and integration to do ( I have only  limited platforms and time ) 

See the current state and plan here: https://github.com/Josverl/pymakr-vsc/blob/fix/SerialMultiPlatform/multiplatform%20support.md 

a strech goal would be to come up with a way to also include support for platforms for which serialport does not publish pre-build, though inclusion or just-in-time building and moving the binding to the correct sub-folder.

OxygenLithium, 2019-08-20:
Is there a fix for windows? I'm still getting this issue, despite the fix back in July

duncanellison, 2019-08-27:
The instructions from GC-RnD above worked for me, with one exception.
My user folder in windows is Duncan<space>Ellison, so I had to manually CD to the folder and use 'Duncan Ellison' to enter my user directory, seems like Powershell doesn't like the space.

closed 2019-09-04 by RalphHogenbirk

Sn3akyP3t3, 2019-10-18:
I had some trouble finding the solution from all the noise posted above.  I found this to be clear and concise about why this has happened and what to do about it.  https://lemariva.com/blog/2018/12/micropython-visual-studio-code-as-ide  Hope it helps others that stumbled onto this.

GC-RnD, 2019-11-08:
Serial port issue has returned....

[cid:image001.jpg@01D59650.C61AE8A0]

This fix no longer works.....

[cid:image002.jpg@01D59650.C61AE8A0]

Help please!

Thank you,
Rich Golden

ttadam, 2019-11-09:
This issue is here again for me also. Is there any known solution/workaround for it?
(Guys I need to make some progress with my work, and don't want to suffer with debugging the development tool, that shouldn't be my job.)

Sn3akyP3t3, 2019-11-10:
I was having other issues and bailed on VSCode to try out VSCodium.  This bug has no longer affected me with a vanilla install.  I uninstalled VSCode and went with VSCodium as a drop-in replacement.

This is the about info:
Version: 1.40.0 (system setup)
Commit: 86405ea23e3937316009fc27c9361deee66ffbf5
Date: 2019-11-08T10:29:58.209Z
Electron: 6.1.2
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.18362

sergiuszm, 2019-11-10:
The fix does not longer work. I did not have any luck with VSCodium either - the same issue.

Josverl, 2019-11-10:
See https://github.com/pycom/pymakr-vsc/issues/62#issuecomment-552193813 for the latest solution that **may** get you unblocked.

tf131, 2021-05-06:
got exactely the same issue today, unfortunately no hack helped me out. Does anyone has a solution in 2021? Thanks!

Josverl, 2021-05-06:
@tf131 I suggest opening a new issue with the current pymakr and VSCode details

moefear85, 2021-08-07:
I'm having this issue on ubuntu 20.04 latest software updates, with vscode v1.56 & pymakr v1.1.12

any hints?

rdagger, 2021-08-07:
You can temporarily downgrade [VS Code to June 2021 (version 1.58)](https://code.visualstudio.com/updates/v1_58) to get pymakr working.  Hopefully, a fix will be released.

GC-RnD, 2021-08-12:
This is like my third or fourth time with VSC update breaking Pymakr.
Can anyone recommend another REPL that works with ESP32 and Visual Studio code.

jalna7, 2021-08-16:
Never had it or noticed it before 
I upgraded to 1.59.0 ( from 1.57.1 ) yesterday and ran into the same problem.  I had also upgrade to the latest version of NodeJS
Have downgraded vsCode and now working again.

PR #57 (2019-09-04): fix(linux): Update electron/serialport for vscode 1.36
Compatibility fix.

(Hi! 👋 Thanks for sending a pull request! Please make sure you click the link above to view the contribution guidelines, then fill out the blanks below.)

## What does this implement/fix? Explain your changes.

* Recompiled `bindings.node` for linux, as discussed in #53 
* Exact Steps:
```sh
yarn global add prebuild-install
cd ~/.vscode/extensions/pycom.pymakr-1.1.3
cd node_modules/@serialport/bindings

prebuild-install --runtime electron --target 4.2.5 --tag-prefix @serialport/bindings@ --verbose --force

cd ~/.vscode/ [... cut]
commit 879331e6 (2019-09-04): fix(linux): Update electron/serialport for vscode 1.36

Compatibility fix.

Closes #53