pycom/pymakr-vsc #55: 🚀 Enhanced colors displayed in the REPL

sebastienkrut, 2019-07-23:
# Feature Request 🚀: Enhanced Colors displayed in the REPL
Currently the PyMakr REPL is missing some shade of gray.
E.g. displaying the same grey text:
- under PyMakr REPL:
[image]
-  under Putty:
[image]
Can PyMakr REPL color management be enhanced?

Josverl, 2019-07-23:
Hi Sebastian, AFAIK the terminal is provided by vscode, and may also be dependant on the OS support.

Can you share your vscode and os details, and that test script to provide a repro?

That would make looking into this a lot simpler.

sebastienkrut, 2019-07-23:
Hi Jose. I am using VSC on Windows and the latest version of Pymakr (after following your instructions to make it work). But I remember that the issue was the same with previous versions.
- VSC
Version: 1.36.1 (user setup)
Commit: 2213894ea0415ee8c85c5eea0d0ff81ecc191529
Date: 2019-07-08T22:59:35.033Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.18362
- Pymakr
v1.1.3

You can use this test code:
```
print("\x1b[1;47m")
print("This white on grey text should be visible")
print("\x1b[0m")
print("Back to normal grey on black text")
```
Thank you for your quick answer.

Josverl, 2019-07-25:
Thanks for the scrip , I can repro that the above script executed on MicroPython results in white on white color mapping,  
while the same script in Python is grey-on-white 
[image]

same version details: 
```
Version: 1.36.1 (user setup)
Commit: 2213894ea0415ee8c85c5eea0d0ff81ecc191529
Date: 2019-07-08T22:59:35.033Z
Electron: 4.2.5
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.18362

Pymakr : 1.1.3
```

sebi5361, 2019-09-14:
@RalphHogenbirk: Can I get an update if this feature will be worked on? Distinguishing grey colors in the REPL is an important feature for me with some of my scripts.