pycom/pymakr-vsc #58: permanent Fix for the serialport problem

Josverl, 2019-08-12:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
see issues #4 #11 #28 #42 #53 #54 
when vscode updates to an electron version that has a different ABI, the serialport module breaks 
these updates seem to come unexpected

## Describe the solution you'd like
during packaging add the native bindings for : 
- multiple platforms 
- multiple electron versions ( ABI versions more specifically) 
this can be done by a script ( approx 200 lines of ps1 code) 

- test against the insiders build of vscode to pre-empt version changes , and publish updates containing updated native bindings, before they are rolled out to the general public 

this would need to be done by an automated build/test system 
the reference uses Azure Pipelines which is free for open source projects such as pymakr.

This is described in the reference implementation documented at :
 https://github.com/Josverl/pymakr-vsc/blob/fix/SerialMultiPlatform/doc/multiplatform%20support.md#multiplatform-support

## Describe alternatives you've considered
see the discussion on #53 

im working on a [PR](https://github.com/Josverl/pymakr-vsc/tree/fix/SerialMultiPlatform) , but as it is quite large, may need to work to bring it into shape.

GeirFrimann, 2019-10-29:
Any news about the PR?

Josverl, 2019-10-30:
The PR is there for three months,
The issues with the serial port library pubkishing were reported to fixed 2 weeks ago.
Waiting for Pycom to take action

Josverl, 2019-11-12:
getting integrated, so closing this issue

closed 2019-11-12 by Josverl

GeirFrimann, 2019-11-12:
Great news @Josverl, tanks!