pycom/pymakr-vsc #60: command 'pymakr.globalSettings' not found

stedesDev, 2019-09-05:
Hi how is life...

Working on win 10 and VS Code 1.38

Did install PyMakr through the extention market place
No command is found... seem like it doesn't see the installation
But seem to be install in the good place with the other extension
C:\Users\blablabla\.vscode\extensions

If any can help, would be very welcome...

It make a few evening searching... and nothing

Best Regards,

Steph

dwjbosman, 2019-10-13:
I'm running on Linux and VSCode 1.39 and have the same issue.

jncolin, 2019-11-08:
Same for me: VSCode 1.40 on Linux Fedora 31
$ node -v
v12.10.0

Josverl, 2019-11-12:
Hi, can you please share the VSCode and Pymakr version details ?
Vscode > Help > About [Copy] 
Pymakr version ?

jncolin, 2019-11-13:
I moved from Fedora 31 to Mint 19.2 and now it's working, with 
Version: 1.40.0
Commit: 86405ea23e3937316009fc27c9361deee66ffbf5
Date: 2019-11-06T18:14:08.920Z
Electron: 6.1.2
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 5.3.0-22-generic

PyMakr 1.1.5

jncolin, 2019-11-13:
Actually, it's not working neither; i get the REPL prompt in VSCode, but when i try to upload the project, I always get the following:
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to /flash...
Reading file status
Failed to read project status, uploading all files
Creating dir lib
[1/10] Writing file accel.py (0kb)
[2/10] Writing file essai1.py (1kb)
[3/10] Writing file lib/LIS2HH12.py (6kb)
[4/10] Writing file lib/LTR329ALS01.py (3kb)
Failed to write file, trying again...
Failed to write file, trying again...
timeout

Josverl, 2019-11-13:
ok , that seems to be a different problem than in the original issue/subject.

we would need to cranck up the logging to see the details; 
but I suspect : 
- some message/background logging from the board interfering with the upload. ( generally this will cause a retry , and then the upload will recover, but if there are many/persistant messages ive seen this happen) 
- the board running out of flash to store files ( error handling should normally catch this) 
- the board running out of RAM (upload is compiling/executing dynamic code as part of the upload)

jncolin, 2019-11-13:
Reagrding running out of RAM or flash, how could that be? It's only uploading small files and fails in the first?
I flashed the latest firmware on pysense and wipy; I tried several usb ports, but still have the problem. 
Also, when at the prompt, if I type: import os and then os.mkfs('/flash'), I get the following error:
pycom AttributeError: 'module' object has no attribute 'mkfs'

even after flashing the new version

pedroknup, 2019-11-14:
Hi @jncolin, is your "safe_boot_on_upload" config in Pymakr > Global Setting enabled? If not, give it a try.

pedroknup, 2019-11-14:
> New file system type is added: Little FS
> Virtual File System concept is used
> The /flash can be LittleFS or FatFS, the SD Card can be FatFS
> Files System related functions are replaced, they are used via VFS
> **mkfs() API is removed and is replaced by fsformat()**

https://forum.pycom.io/topic/5258/firmware-release-v1-20-1/56

jncolin, 2019-11-18:
> Hi @jncolin, is your "safe_boot_on_upload" config in Pymakr > Global Setting enabled? If not, give it a try.

Hi, this actually did the trick; I could now upload my project and run it! Thanks

stedesDev, 2019-11-21:
Hi....
Anything new for the first comment which is not the one been solved...
I try and retry... nothing work... will send you more installation once at home...

pedroknup, 2019-11-21:
Which version of Pymakr do you have?

stedesDev, 2019-11-22:
Hi,
How is life…

PyMark version is  1.1.5…
It install well… see all the things in the bar… try to connect and always no device found…
In the config file I change for the good COM port…

Did install micropy-cli.. every things is OK…

Whit the upyCraft IDE every things work fine… it see my  board can download to it…

Regards,

stedesDev, 2019-11-22:
I have a ESP32 WROOM 32 from Keyestudio… and they tell to select it like the ESP32 Dev Module…

Regards

stedesDev, 2019-11-22:
Hi…
Can it be in conflict with the Arduino extension… I disable the Arduino extension and seem to work…

Good good good

Thank a lot for your time and patient

Regards,

Steph

closed 2020-01-02 by pedroknup