pycom/pymakr-vsc #61: Receiving the error while installing the paymakr on atom

lodhi830, 2019-10-13:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What happens?
I am receiving the errors that There was an error with your serial port module, paymakr will likely not work properly. ....

[image]

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## What versions of the software are you using?

**Operating system:** Ubuntu 18.10

**Pymakr version:**  1.4.17

elmasria, 2019-10-21:
Hello @lodhi830, please try to remove the whole **pymakr** package, then install the latest **1.4.17**, see if that solves the issue.

Josverl, 2019-12-31:
I assume this should be closed or moved/copied to pymaker for Atom

pedroknup, 2020-01-02:
This issue has been fixed with the latest Pymakr Atom update.

closed 2020-01-02 by pedroknup