pycom/pymakr-vsc #74: Network connection support for ESP / webrepl?

metachris, 2020-01-03:
How would I connect Pymakr to a generic ESP32 micropython device on the network? Does it work with webrepl?

Josverl, 2020-01-16:
It works with a full telnet server, which is available in the firmwares from Pycom an Lobo ESP32. I have not found a decent working telnet server for std Micropython, but there may be one around. It wound need to allow for authentication though

rdagger, 2020-09-12:
@Josverl Did you ever find a telnet server for std Micropython?  Or is there any way to make it work with FTP?  The serial connection is slow.

Josverl, 2020-09-14:
@rdagger , not one that works well with pymakr. the usual lack is that they do not implement authentication , and pymakr just keeps waiting for the auth prompt.
it should not be too hard to add a dummy user / password prompt though , but i've never completed that. 
the one I briefly looked at is indeed  [cpopp/MicroTelnetServer](https://github.com/cpopp/MicroTelnetServer)

I if you can add 2 prompts for  `Login as:`  and `Password:` and just ignore whatever is put in there you may be good to go .
[image]

another option is to create a PR for pymakr so that it does not need to put in the user/pwd ( ie if you spefify them as empty ) or by detecting the utelnetserver prompt : 
```
Connected to esp_f4b4b3.
Escape character is '^]'.
```

FTP:  i never liked that even with the LoBo firmwares. there were to many moving parts that got tangled most of the time.

rdagger, 2020-09-14:
@Josverl Thanks for your reply.  I’ve found that FTP works great for transferring files.  It’s an order of magnitude faster for transfers compared to serial.  I use the following light-weight FTP server:  https://github.com/robert-hh/FTP-Server-for-ESP8266-ESP32-and-PYBD

I was using a plugin in Atom to automatically upload my code to the ESP32 via FTP.  It works well except when it comes to running the code which requires a separate shell and manual input.

Pymakr does a great job of streamlining the process.  Unfortunately, the serial transfers are tedious for larger libraries.
The question may be moot because it looks like Pymakr is no longer compatible with the latest version of VS Code and the project hasn’t had any development for 6 months.

rdagger, 2020-09-19:
@Josverl 
> it should not be too hard to add a dummy user / password prompt though , but i've never completed that.
>the one I briefly looked at is indeed cpopp/MicroTelnetServer

I implemented the dummy credentials prompt and it worked.  Pymakr can now connect via telnet.  Unfortunately, the file upload doesn't work.  I get an error before the ESP32 reboots:

> raw REPL; CTRL-B to exit
> >OK'/'
> >OKW1siZm9udHMiLCJkIiw ...
> 
> >OKTraceback (most recent call last):
>   File "<stdin>", line 4
> SyntaxError: invalid syntax
> >OKTraceback (most recent call last):
>   File "<stdin>", line 4
> SyntaxError: invalid syntax
> >OKTraceback (most recent call last):
>   File "<stdin>", line 4
> SyntaxError: invalid syntax
> >OKI (523) cpu_start: Pro cpu up.
> I (523) cpu_start: Application information:

I suspect MicroTelenetServer doesn't implement a required command.

Josverl, 2020-09-30:
@rdagger , the upload / download essentially uses the interactive prompt to create files on the fly,
so if comms work , that should work as well.

do you have a pointer to your dummy creds prompt ?
ill be happy to trace what throws a cog into the wheels.

rdagger, 2020-10-12:
@Josverl All I did was add the following lines to the _def accept_telnet_connect() method_.

```
    last_client_socket.setblocking(True)
    last_client_socket.write("Login as:")
    last_client_socket.readline()
    last_client_socket.write("Password:")
    last_client_socket.readline()
    last_client_socket.setblocking(False)
    last_client_socket.write('Login succeeded!\r\nType "help()" for more information.\r\n')
```

metachris, 2020-11-05:
How do you connect to the telnet server, as in which pymakr `address` setting are you using?

rdagger, 2020-11-06:
@metachris I set address to the IP address in the global pymakr.json settings file.  I don't know if you can specify a custom port.