pycom/pymakr-vsc #75: Ubuntu. NodeJS not detected on this machine!

hcmila, 2020-01-04:
Version: 1.41.1
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T15:04:31.999Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 4.15.0-72-generic snap

pedroknup, 2020-01-06:
You need to have NodeJS installed in order to run Pymakr on Visual Code. Assuming that is your Node version, can you make sure you can run some node commands? node -v for example.

hcmila, 2020-01-06:
As I wrote before I have node installed on my ubuntu system:
~$ node -v
v13.0.0-nightly20190802452b393c1f
Just after start vsc there is New Notification:
NodeJS not detected on this machine, which is required for Pymakr to work. See the Pymakr readme for dependancies.

kbondarev, 2020-05-25:
I am experiencing the same issue and I have NodeJS installed:
```bash
$ node --version
v14.3.0
```

pymakr extension version: `1.1.6`

VS Code information copied from the About window:

```txt
Version: 1.45.1
Commit: 5763d909d5f12fe19f215cbfdd29a91c0fa9208a
Date: 2020-05-14T08:27:22.494Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 5.4.0-7629-generic snap
```

kbondarev, 2020-05-25:
Update:

I managed to resolve this. The issue seemed to be that my nodejs was installed as a snap package and my guess is that for some reason vscode could not find it on the PATH. 

There is probably a way to make it work with the snap version of node, but I ended up just removing the node snap and installing it through the [package manager](https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions-enterprise-linux-fedora-and-snap-packages)

pedroknup, 2020-05-25:
Thanks for the update, @kbondarev

Schade77, 2020-06-08:
Hello, 

I get some similar trouble, with vscode 1.45.1, nvm which different node version tested. Command node -v well working on my terminal. 
After to install, Pymakr, I get the error message : "The terminal shell path "node" is a directory. I thinks something is wrong on my node installation or path definition with nvm.

It's working great on my laptop with apt-get nodejs. 

Any ideas ? 

Regards

ljmerza, 2021-01-08:
same issue here. i use nvm so do not want to install through package manager. This is an issue with pymakr as all other apps can detect nodejs.

closed 2021-05-17 by elmasria