pycom/pymakr-vsc #76: Upload single file to board

Jerremy09, 2020-01-06:
Hello,

is there a reason why single file upload to board do not work for ESP32?  I can upload whole project sucessfully, but I cannot upload a single file.

My settings:
VS Code:
Version: 1.41.1 (user setup)
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T14:58:56.166Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.16299

Pymakr: 1.1.5

Board: Lolin D32 Pro V1.0.0

Error: "Failed to write file" - this error is listed multiple times in Terminal.

Jerremy09, 2020-01-06:
Running system: Windows 10 (64)

fdelmotte, 2020-04-06:
> Running system: Windows 10 (64)

Hello, Do you have any feedback, I have the same issue with an ESP8266 :(

Jerremy09, 2020-04-10:
Hello,

unfortunately not, but I found that you can run function "Update" and PYMAKR recognize changes and upload only changed files. That Is the situation with which I'm fine.

fastovenko, 2021-11-07:
I have the same issue. Does anyone know how to fix it?

closed 2022-03-24 by VASKOTechDesign