pycom/pymakr-vsc #99: Pymakr v1.1.6 doesn't work with VSC latest update 1.49.0

sebi5361, 2020-09-10:
My VSC has just updated to the latest version and now Pymakr refuses to work...
Pymakr v1.1.6
VSC:
Version: 1.49.0 (user setup)
Commit: e790b931385d72cf5669fcefc51cdf65990efa5d
Date: 2020-09-10T13:22:08.892Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19033

bland328, 2020-09-11:
I'm experiencing the same thing. My version numbers are the same, with the exception that I'm on Windows 10 10.0.19041 Build 19041.

I've tried uninstalling, reinstalling, rebooting between those steps, and every trick I can think of.

When I try to get Pymakr to connect to my ESP32 devkit that was previously working, I get this error in Visual Studio Code:

`Command 'Pymakr > Connect' resulted in an error (command 'pymakr.connect' not found)`

I'm happy to gather more data, if it's of use.

yijun-sphere, 2020-09-11:
Same thing here, tried both on macOS 10.15 and ubuntu 20 with the latest vscode. Pymakr was running perfect before I click the vscode update.

Tried to install and uninstall everything back n forth with no luck.

Pymakr terminal is not showing, when using shortcut key to try to execute connect the error pop up says command 'pymakr.connect' not found.

Node: 14.10
Pymakr: 1.1.6

jabelone, 2020-09-11:
I'm on macOS Catalina and I just started experiencing this issue. The same thing even happens if I build and install pymakr from source. If I open the dev console, the following error is logged:

```
[Extension Host] Error: Could not locate the bindings file. Tried:
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/build/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/compiled/12.14.1/darwin/x64/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/lib/binding/node-v80-darwin-x64/bindings.node
	at bindings (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/bindings/bindings.js:126:9)
	at Object.<anonymous> (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/lib/darwin.js:2:36)
	at Module.o._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:360)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1051:10)
	at Module.load (internal/modules/cjs/loader.js:862:32)
	at Module._load (internal/modules/cjs/loader.js:774:14)
	at Function.Module._load (electron/js2c/asar.js:769:28)
	at Function.t._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1046:268)
	at Function.i._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1015:981)
	at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1012:303)
	at Module.require (internal/modules/cjs/loader.js:899:19)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:135)
	at Object.<anonymous> (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/@serialport/bindings/lib/index.js:10:22)
	at Module.o._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:360)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1051:10)
	at Module.load (internal/modules/cjs/loader.js:862:32)
	at Module._load (internal/modules/cjs/loader.js:774:14)
	at Function.Module._load (electron/js2c/asar.js:769:28)
	at Function.t._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1046:268)
	at Function.i._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1015:981)
	at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1012:303)
	at Module.require (internal/modules/cjs/loader.js:899:19)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:135)
	at Object.<anonymous> (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/node_modules/serialport/lib/index.js:2:17)
	at Module.o._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:360)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1051:10)
	at Module.load (internal/modules/cjs/loader.js:862:32)
	at Module._load (internal/modules/cjs/loader.js:774:14)
	at Function.Module._load (electron/js2c/asar.js:769:28)
	at Function.t._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1046:268)
	at Function.i._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1015:981)
	at Function.n._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:1012:303)
	at Module.require (internal/modules/cjs/loader.js:899:19)
	at n (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:16:135)
	at prepareSerialPort (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/pymakr.js:148:9)
	at activate (/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.6/pymakr.js:7:5)
[... cut ...] ench/services/extensions/node/extensionHostProcess.js:917:841)
	at Function._callActivate (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:917:492)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:915:724
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async Promise.all (index 5)
	at async Promise.all (index 0)
```

sebi5361, 2020-09-11:
Just a trick: Till this bug has been fixed, you can downgrade VSC to its previous version (1.48.2) and disable VSC automatic updates making Pymakr work as before...

ReinoutHeeck, 2020-09-11:
I managed to hack this to work, serialport needs to be upgraded to use Node v80.

in package.json set version of
 node-abi to 2.19.1 
 serialport to 9.0.1
update serialport version in scripts/mp-download.ps1 too.

then do 
 npm install
 npm run download-native
run the build task
run the configuration 'Launch Extension'

AkhileshThorat, 2020-09-13:
Facing same issue

imliubo, 2020-09-14:
Same issue...
```bash
sudo autoremove code
sudo apt install code=1.48.2-1598353430
```
**and**
[image]

Manfredi9, 2020-09-14:
Same issue...

brilliantwf, 2020-09-14:
Same issue...

elmasria, 2020-09-15:
Please check the new version 1.1.7

ReinoutHeeck, 2020-09-15:
@elmasria 

New 1.1.7 works for me on win10 - thank you!

sebi5361, 2020-09-15:
Thank you very much for your reactivity @elmasria !
I will close this issue in a few days only to let other people read about the solution in the meantime.

sebi5361, 2020-09-15:
Certainly not related to the new version as I remember I already had this issue in the past
...but Pymakr takes a few automatic iterations to work fine on my computer without throwing such an error message a multiple time in a row:
[image]

Manfredi9, 2020-09-15:
Thank you @elmasria! 
On MacOs works properly.

jabelone, 2020-09-17:
I just built 1.1.7 from the `master` branch and it now works with VS Code 1.49.0 on macOS. Thanks!

closed 2021-05-17 by elmasria