pycom/pymakr-vsc #107: Undeleteable file created after upload

rdehuyss, 2020-11-15:
## What are the steps to reproduce this issue?
1. Create a project with a file 'main.py' and a folder 'main'
2. Upload project

## What happens?
A `py` inode(?) is created in the folder `main` which is un-deletable (it is not a file nor a folder)

## What were you expecting to happen?
To not have a `py` something (to be honest, I don't know what it is) in the folder `main`.

## Any logs, error output, etc?
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> import os
>>> os.listdir()
['boot.py']
>>> 
Uploading project (src)...
Not safe booting, disabled in settings

Uploading to /flash...
Reading file status
Failed to read project status, uploading all files
Creating dir main
[1/2] Writing file main/test.py (0kb)
[2/2] Writing file main.py (0kb)
Upload done, resetting board...
OKets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4880
ho 0 tail 12 room 4
load:0x40078000,len:8652
load:0x40080000,len:5748
entry 0x400802dc

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 2621440; Flash address: 0x180000]
----------------
Filesystem size: 2402304 B
           Used: 2304 B
           Free: 2400000 B
----------------
test
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> import os
>>> os.listdir()
['boot.py', 'main', 'main.py', 'project.pymakr']
>>> os.listdir('main')
['test.py', 'py']
>>> os.chdir('main')
>>> os.listdir()
['test.py', 'py']
>>> os.remove('py')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> os.rmdir('py')
>>> os.listdir()
['test.py', 'py']
```

## Any other comments?
Thank you for this excellent VSCode extension!!!

## What versions of software are you using?

**Operating system:**
Ubuntu 20.04

**VSCode version:**
1.51.1

**Pymakr version:**
v1.1.7

Josverl, 2020-11-16:
Well, [spiffs] [] actually does not support folders or subdirectories.
They are faked by using filenames that include a `/`

I assume that in this case where you have overlapping names between your folder and a file, that the folder emulator cannot cope .

I've seen other issues where, using Micropython, I was able to create files in folders that did not exist, and that only showed up after I created the folders.

In your case I think that your `main.py` file will show up after you remove the `main` folder

Im pretty sure that the current upload code does not try to preempt or correct such overlap, but it would be good to document this just to be sure.

[spiffs] : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spiffs.html

Josverl, 2020-11-19:
Hi Ronald,

i setup a minimal repro using a snippet of your code and can't find the issue you describe 

See https://github.com/Josverl/rmlist-poc

Ill try to add the ota updater, interesting concept 👍.

Groeten uit Holland :-) 

/jos

rdehuyss, 2020-11-19:
Hi Jos,

Mmm, I think I have/had the exact same small repro as you did (I should have attached it in the first place). Will try to set it up again as soon as I have some time. I wonder if it's perhaps related to the version of MicroPython that is runnig (e.g. ESP32_LoBo_v3.2.24) for the M5Stack.

I've only seen the strange file/folder issues on an M5, not on a ESP32 with standard MicroPython.

Groetjes en bedankt om op te volgen!