pycom/pymakr-vsc #108: Project upload does not remove ignored folders with multiple levels.

Josverl, 2020-11-19:
1. upload 2 files in two nested folders : Upload-1 
 - somefolder
  - file1.py
  - subfolder
     - file2.py 
2. change project configuration to block uploading 'somefolder'

``` json
{
    "addresx": "192.168.4.1",
    "username": "micro",
    "password": "python",
    "sync_folder": "",
    "open_on_start": false,
    "safe_boot_on_upload": false,
    "py_ignore": [
        "pymakr.conf",
        ".vscode",
        ".gitignore",
        ".git",
        "project.pymakr",

        "somefolder"
    ],
    "fast_upload": false
}
```
3. Upload again : Upload-2

## What happens?
**Upload-1**
``` 
>>>
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to /...
Reading file status
Creating dir somefolder
Creating dir somefolder/level2
[1/2] Writing file somefolder/file1.py (0kb)
[2/2] Writing file somefolder/level2/file2.py (0kb)
Upload done
```

Upload-2   
**Runs into an error: `Failed to remove dir somefolder`** 
```
>>>
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to /...
Reading file status
Deleting 4 files/folders
Removing file somefolder/level2/file2.py
Removing file somefolder/file1.py
Removing dir somefolder
Failed to remove dir somefolder
Removing dir somefolder/level2
Upload done
```

Somefolder is not removed as is shown below
```
>>> os.listdir()
['clean.py', 'config.json', 'micropython-ota-updater', 'project.pymakr', 'somefolder']
>>> os.listdir('somefolder')
[]  
```

## What were you expecting to happen?
- somefolder to be removed 

## Any other comments?
Possibly related to #107 as i discovered this while researching that

The pymakr javascript upload code does not seem to work well with folders. It seems that it was once written for a single level only.
Ive tried to update it some time ago , but IMO it requires a rewrite 

## What versions of software are you using?
Pymakr: v1.1.7

VSCode
Version: 1.51.1 (user setup)
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-10T23:34:32.027Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19042

SebastiaanMerckx, 2021-11-05:
This issue might rather be related to the order of execution.
'somefolder' is removed before 'somefolder/level2'. At the moment of removing 'somefolder', it is not empty and hence it fails (just like a "rm directory" command fails in linux when the directory has a subdirectory). Changing the sequence might fix it.

Not sure though, since I don't know the filesystem on micropython is handled exactly.

Josverl, 2021-11-05:
im pretty sure the issue is not in how MicroPython handles the deletion of non-empty folders, but in the fact that the Pymakr file handling code appears to be originally written for a flat file system, and folders were added later ( see above) 

i think a large section of js(ts?) code can be replaced by a not too complex recursive (embedded) micropython script that traverses folder & files, creates a crc32 and returns/writes this as json to `project.pymakr` .
(i'm pretty sure I shared such a script at one time or the other) 

Then finding the need for , and making the changes is as simple as comparing the new and current json tree in either js or python , and traverse though the changes  in a Depth-first search manner.