pycom/pymakr-vsc #115: Raspberry Pi Pico support

sebi5361, 2021-02-01:
It seems there is no possibility to copy files using the Pymakr vsc addon on windows.
Is there any chance you want to fix that?
Other tools such as **rshell** encountered similar issues and had to rely on small tweaks to make it work:
https://github.com/micropython/micropython/issues/6816
[@dpgeorge](https://github.com/dpgeorge) believes it could be a problem with flow control (at the REPL/raw REPL level), and says that using the new raw paste mode from the latest pyboard.py might fix the problem. Is the pymakr vsc addon built on top of this new pyboard.py file?

Josverl, 2021-02-08:
@sebi5361,
pymakr uses the [node serialport](https://github.com/serialport/node-serialport/)  library which is used widely 
as there are a few more puzzling and outstanding issues on the micropython/rp2 port , this seems to indicate that hte issue is somewhere in the usb/serial handling on that hardware / port .

that being said, if you have repro steps then i'm happy to take a look at this

sebi5361, 2021-02-08:
Indeed the issue seems to come from the rp2 port itself, and will be certainly fixed in the coming days.
I tried to experiment again with Pymakr to give you repro steps but I think I am facing the same issues as in https://github.com/pycom/pymakr-vsc/issues/119 since vscode did update in the meantime.
Anyhow, from what I recall, the issue occured when trying to synchronize a local project folder with the board: I got some error messages saying files couldn't be copied, and after a few attempts, the synchronization failed.
Regarding the current issue with Pymakr https://github.com/pycom/pymakr-vsc/issues/119, if I remember well we already encountered such an issue in the past, and you came out with a temporary solution to overcome it, as well as a suggestion (or even a pull request) to fix such issues permanently. You also submitted proposals for improvements like remote folder overview. I don't know the exact situation at Pycom, but from my understanding you are not involved there directly, being though the most competent and responsive in maintaining this software. Any chance you want to start your own fork?

Josverl, 2021-02-09:
@sebi5361  
I'm not sure about the priority given to pymakr either - seems that it is mostly in a reactive break-fix mode , although whenever i had direct interactions that was really OK.

I have no relation to Pycom, (but I do have an association to VSCode ...) 
I've been thinking about forming as well, but would not want to do this on my own, and would rather work together than build a something in isolation.

sebi5361, 2021-02-09:
I think it would be nice.
It's a shame the Pymakr-VSC addon is not among Pycom's priorities anymore.
Unfortunately I am not skilled enough in programming to help you with this aspect, but I can do reviews, write some descriptions and report bugs.
Anyhow with Raspberry entering the Micropython game and as there are not that many good Micropython addons for VSC, I am sure you might bring a nice community around your project.
Just a thought...

cpwood, 2021-02-15:
Hi all - take a look at https://github.com/cpwood/Pico-Go

sebi5361, 2021-02-15:
I didn't try it yet but it looks really promising to me!
- It seems to me you only want to tackle the Pico board. Is that true?
- Do you want to bring additional features (like code stubs...) to the addon or do you prefer to stay minimal?

cpwood, 2021-02-15:
Yes - at least just the RP2040 boards. I'd be spread a bit thin taking on others. Plus Pymakr has already got PyCom covered, etc.

Stubs are [available here](https://github.com/cpwood/Pico-Stub) and are in constant refinement. They'll be included _in_ the Pico-Go extension from the next minor version very, very soon. You'd just need to choose "Configure project" from the Command Palette and it would automatically configure everything, though there's nothing to stop anybody using Micropy-CLI too.

sebi5361, 2021-02-15:
There is something that Pymakr doesn't do: it is to connect to wifi-enabled devices (ESP8266,  ESP32, ...) via the WebREPL.
This would be neat and would allow to program those boards wirelessly.
[Zepl](https://gitlab.com/zepl1/zepl) is a simple remote console application for Micropython webREPL, with file transfer capabilities ([pip install zepl](https://pypi.org/project/zepl/) to install). Integrated to your addon, this would provide a transparent user experience when dealing with  such boards wirelessly.
Is such a feature for Pico-Go conceivable?

cpwood, 2021-02-15:
@sebi5361 not conceivable for the moment, no: the Pico doesn't have a WiFi capability. It's serial port only.

Josverl, 2021-02-16:
@sebi5361, pymakr dos serial and __some__ telnet connections (these forked from loboris) .
expanding to Any telnet , and adding the webrepl or ftp protocols would be quite useful indeed.

sebi5361, 2021-02-22:
@Josverl I agree with this, as special firmwares are needed to use the telnet feature.
Using [MicroTelnetServer](https://github.com/cpopp/MicroTelnetServer) and [uftpd](https://github.com/robert-hh/FTP-Server-for-ESP8266-ESP32-and-PYBD) or the webrepl with the Pymakr addon would be a must!

roaldarbol, 2021-12-17:
Hi all! Is there any progress on this issue? I can see from the @cpwood 's `Pico-Go` that he doesn't have time to maintain it (that's correctly understood right @cpwood?), so would be great to know that there is continued support and development for the Pico.

MySurmise, 2022-08-21:
Please approach this issue. I am also using `Pico-Go`, and although it could be a great peace of software, without an active maintainer it's useless. I currently have a bug that makes it impossible to communicate with my pico w and because no issues/PR are accepted, the github repo is dead. The pico w has great potential and pymakr could help unleash that!

Josverl, 2022-08-23:
@MySurmise, 
can you please add the bug details so other can understand what the issue might be ?

I just 
- connected my RP Pico to USB 
- in Pymaker 2 
  - connected the device 
  - added it to an existing project 
  - pressed upload 
    ( all project files were uploaded ) 
  - hover over the connection shows info of the RP firmware and filesystem
  - I can open a repl prompt and interact with it 

Conclusion: 
to me it appears that the majority of the features are supported, and this issue should be closed.
if there are specific additions that could be useful, then I suggest to start a [**Discussion** ](https://github.com/pycom/pymakr-vsc/discussion) to clarify what you are after

jakobrosenberg, 2022-08-23:
Thanks for the feedback @Josverl and @MySurmise . I will close this as solved.

closed 2022-08-23 by jakobrosenberg

sgbaird, 2022-11-04:
There's [a Pico-W-Go extension](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go)