pycom/pymakr-vsc #116: Vscode 1.53.0 update on osx breaks pymakr

buffedelic, 2021-02-05:
The extension stopped working after vscode 1.53.0 update on osx.

`Command 'Pymakr > Connect' resulted in an error (command 'pymakr.connect' not found)`

PyMakr version 1.1.7

elmasria, 2021-02-05:
Hi @buffedelic,

Thank you for reporting this we will try to fix it asp. 

Best Regards,
Ahmad EL Masri

GargamelCZ, 2021-02-06:
Great. :-D I didn't work with it for quite some time so I had no idea it's related to this, so I've spent all day debugging everything else. Only thin I've found related to this is that Pymakr only sees the default bluetooth usb and nothing else:
`Found 1 serialport`
`/dev/tty.Bluetooth-Incoming-Port (undefined) (copied to clipboard)`

ZorroWasHere, 2021-02-07:
> Great. :-D I didn't work with it for quite some time so I had no idea it's related to this, so I've spent all day debugging everything else. Only thin I've found related to this is that Pymakr only [...]

You need to add manufacturer of your comport chip in global settings, like:
"autoconnect_comport_manufacturers": [
		"Silicon Labs",
		"Pycom",
		"Pycom Ltd.",
		"FTDI",
		"Microsoft",
		"Microchip Technology, Inc."
	]

Josverl, 2021-02-10:
Meanwhile to get you unstuck you can consider using a beta version from my fork : 

https://github.com/Josverl/pymakr-vsc/releases/tag/v1.1.7-beta
Download and  install a .vsix by opening the file from within VS Code. 
Run File | Open File... or kb(workbench.action.files.openFile) and select the extension .vsix.

later you can uninstall an re-join the pymakr mainline.
HTH  Jos

elmasria, 2021-02-16:
Hi all,

We have deployed a new version 1.1.9 which should solve the issue.

Thank you for your feedback

Best Regards,
Ahmad El Masri

closed 2021-05-17 by elmasria