pycom/pymakr-vsc #118: Upgrade VScode 1.53 , could not connect

fredvol, 2021-02-06:
Hello ,

After an the auto update on vs code 1.53, the pymakr extension could not connect to the board.
I could connect to the board using:
- arduino IDE serial port
- screen  in the commans line 

My settings:

- linux mint 20
- Tested on Esp8266  , ESP12E  and ESP32
- Vs code: 
        Version: 1.53.0
        Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
        Date: 2021-02-03T15:56:04.185Z
        Electron: 11.2.1
        Chrome: 87.0.4280.141
        Node.js: 12.18.3
        V8: 8.7.220.31-electron.0
        OS: Linux x64 5.4.0-65-generic

I uninstall and reinstall pymakr  for the extension tab hoping to get the last version of pymakr v1.1.8 , but still not working.

Then install Codium on the side , here the detail version :
          Version: 1.52.1
          Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
          Date: 2020-12-17T00:37:39.556Z
          Electron: 9.3.5
          Chrome: 83.0.4103.122
          Node.js: 12.14.1
          V8: 8.3.110.13-electron.0
          OS: Linux x64 5.4.0-65-generic

and it was working fine again.

Thanks for your work.
Fred

cessna182j, 2021-02-06:
fredvol - how did you install pymakr in VSCodium?  Tried vsix install and "not compatible".

fredvol, 2021-02-07:
I follow that : 
[https://github.com/VSCodium/vscodium/blob/master/DOCS.md#extensions-marketplace](https://github.com/VSCodium/vscodium/blob/master/DOCS.md#extensions-marketplace)

It install and work fine on Vscodium (V 1.52.1)
It seems to me that the problem in on the upgrade of vscode 1.53.

elmasria, 2021-02-16:
Hi all,

We have deployed a new version 1.1.9 which should solve the issue.

Thank you for your feedback

Best Regards,
Ahmad El Masri

fredvol, 2021-02-16:
Yes works for me on linux  with pymakr 1.1.11.
Thanks for your work.

closed 2021-02-16 by fredvol