pycom/pymakr-vsc #119: VS Code update - serial not working 1.53.0, Electron 11.2.1

cessna182j, 2021-02-06:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. Start VSCode with Pymakr 1.1.8
2. LoPy module connected to USB/serial port
3.

## What happens?
> Failed to connect (Error: Port is not open). Click here to try again.

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
VS Code updated then serial no longer working - used less than 24 hours before VSC auto update and no issues.  Had this issue back in 2019 and resolved by updating serial bindings using a prebuild install.  

## Any other comments?
tried previous fixes to update bindings and no success

## What versions of software are you using?
1.53.0
**Operating system:**
Win 10
**VSCode version:**
1.53.0
**Pymakr version:**
1.1.8

arstorey, 2021-02-06:
I'm having the same issue on the same OS and version numbers of VSCode & Pymakr.  I'm running node.js version 14.15.4.  This particular combination of versions has issues both enumerating and accessing COM ports under Windows 10.

cessna182j, 2021-02-06:
Tried VSCodium, installed Pymakr 1.1.8 from vsix - same issue:  Failed to connect (Error: Port is not open). Click here to try again.

Version: 1.53.0 (system setup)
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-06T23:06:26.959Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041

somervda, 2021-02-07:
I have the same issue with 1.1.8 "Failed to connect (Error: Port is not open). Click here to try again." Note: the actual Serial port is working because I can still connect via putty, also pycom->Extra->List serial ports still works.
OS: Windows 10 Pro - 19042.789 
Node: v12.9.1

Lars-O-Knudsen, 2021-02-07:
I also have this issue with v1.1.8, Failed to connect, port not open. 

When I try to use the "Install other version" under extension, it seems to install, but Pymakr is not working at all, ie "pymakr.listCommands" not found.

Here is info from vsc/about:

Version: 1.53.0 (user setup)
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T20:36:38.611Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041

ehtnevets, 2021-02-09:
AutoConnect enabled, ignoring 'address' setting (see Global Settings)
Searching for PyCom boards on serial...
Connecting to /dev/tty.usbserial-0001...
> Connection timed out. Click here to try again.

Version: 1.53.0
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T16:01:04.046Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 20.3.0

danalvarez, 2021-02-09:
Same issue here. There appears to be a pull request pending review that might fix this: https://github.com/pycom/pymakr-vsc/pull/117

Please @Xykon look into this.

elmasria, 2021-02-09:
Hi Guys, the issue is with the Serialport library, which has some dependencies that throw errors on electron 11.

@danalvarez Yes we are working on that PR to fix the issue but did not finish yet. 

Thank you

Josverl, 2021-02-10:
Meanwhile to get you unstuck you can consider using a beta version from my fork : 

https://github.com/Josverl/pymakr-vsc/releases/tag/v1.1.7-beta
Download and  install a .vsix by opening the file from within VS Code. 
Run File | Open File... or kb(workbench.action.files.openFile) and select the extension .vsix.

later you can uninstall an re-join the pymakr mainline.
HTH  Jos

Lars-O-Knudsen, 2021-02-11:
Hi Jos,

Thanks for helping out with a quick fix.
I tried to download the .zip file, but I didnt find a .vsix file included. Could you please elaborate a bit on the instructions?

Br Lars

ehtnevets, 2021-02-12:
I managed to create .vsix file (using vsce package) by following the instruction from this git but vsce doesn't create the lib folder. If I open the project folder with vscode and use task build, it does create the lib folder however the vsix package doesn't include this lib when installing the extension. Back to rshell until a working package is published.

Josverl, 2021-02-12:
You should be able to just download the VSIX,
[image]

elmasria, 2021-02-12:
Hi @Josverl,

The provided package did not load correctly.

[image]

Best Regards,
Ahmad EL Masri

Lars-O-Knudsen, 2021-02-14:
Hi Jos
I have installed the .vsix file, but I get errors.
Please find the log attached.
Br Lars
[pymakr.log](https://github.com/pycom/pymakr-vsc/files/5976784/pymakr.log)

ehtnevets, 2021-02-15:
Somebody on the other thread mentioned pico-go. I tried it and it works just like pymakr used to be.

somervda, 2021-02-15:
Thanks - pico-go worked for me and will keep me going until pymakr is back.

elmasria, 2021-02-16:
Hi all,

We have deployed a new version 1.1.9 which should solve the issue.

Thank you for your feedback

Best Regards,
Ahmad El Masri

littlebell, 2021-02-16:
Hi ,
I hav upgraded to pymakr 1.1.19  but it's still doesn't work 
This is my configuration
Linux Mint 19.3
VScode  1.53.2
Electron 11.2.1
Now Pymakr doesn't start (with previous  1.1.18 pymark start but don't find any serial port) and give me this Message:
'There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)'.
Picocom works well.
Any suggestion?

somervda, 2021-02-16:
Same here - 1.19 still gives this error for me:
There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)

Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

Josverl, 2021-02-16:
@elmasria ,
I think there is a mismatch in the windows binding, 
\node-v85-win32-x64\bindings.node is not a valid Win32 application.

Help > Delveloper Tools > 

`Error: \\?\c:\Users\me\.vscode\extensions\pycom.pymakr-1.1.9\node_modules\@serialport\bindings\lib\binding\node-v85-win32-x64\bindings.node is not a valid Win32 application.
\\?\c:\Users\me\.vscode\extensions\pycom.pymakr-1.1.9\node_modules\@serialport\bindings\lib\binding\node-v85-win32-x64\bindings.node`

littlebell, 2021-02-16:
> Hi ,
> I hav upgraded to pymakr 1.1.9 but it's still doesn't work
> This is my configuration
> Linux Mint 19.3
> VScode 1.53.2
> Electron 11.2.1
> nodejs/bionic-updates,now 8.10.0~dfsg-2ubuntu0.4 amd64 [installed]
> Now Pymakr doesn't start (with previous 1.1.8 pymark start but don't find any serial port) and gives me this Message:
> 'There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)'.
> Picocom works well.
> Any suggestion?

I Solved the binding problem now pymakr start but now i have the same problem of 1.1.8 version :
'Failed to connect (Error: Port is not open)'
Picocom works well.

elmasria, 2021-02-16:
A new version 1.1.10 wich should solve the issue for windows OS

@littlebell, can you give me more details maybe try to open the dev tool and go to the console. and share the details. 

Best Regards,
Ahmad El Masri

littlebell, 2021-02-16:
This is my console output:
[image]

elmasria, 2021-02-16:
@littlebell 

Thank you for the above, but can I ask you to take a screenshot for the start of the error so I can have a better view.

Maybe it is better to go to 1.1.10 also and then test 

Best Regards,
Ahmad EL Masri

littlebell, 2021-02-16:
Ok this should be all output:
[image]
[image]

elmasria, 2021-02-16:
Hi @littlebell can you check version 1.1.11, please

littlebell, 2021-02-16:
Hurray, it finally works again with the new version 1.1.11. I'm happy !
Thank you very much elmasria, you have been patient and kind.

elmasria, 2021-02-17:
@littlebell 

I an m glad that this issue is loved. In the end, I am doing my job :) 

Yeah, I am interested to know more about your projects.

Have a nice day.

Ahmad EL Masri

cessna182j, 2021-02-17:
1.1.11 works!  Nice!
Is the general sense that this fix is "permanent", i.e. is it likely that serial-connect will fail at next electron/VSCode update?

Tarun75, 2021-02-21:
hi, I am facing error with oymakr on VScode, below is the detail, I am working on Rpi4 debian OS

****VS code**** 
Version: 1.51.1
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-10T23:30:36.006Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Linux arm 5.4.51-v7l+

**Pymakr version**
Version:1.1.11

Error:There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)
I am attaching the log from VS code, please help solve the issue. As I am beginner will request if youn can provide step wise process.

[pymakr.log](https://github.com/pycom/pymakr-vsc/files/6017831/pymakr.log)

jwoolaway, 2021-08-16:
Hello, I am havng the same error as many others with Pymakr on VScode, I am working on OSx 11.5.1 (20G80)

Version: 1.59.0
Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
Date: 2021-08-04T23:14:40.191Z (1 wk ago)
Electron: 13.1.7
Chrome: 91.0.4472.124
Node.js: 14.16.0
V8: 9.1.269.36-electron.0
OS: Darwin x64 20.6.0

Tried multiple Pymakr versions
Version:1.1.10, 1.1.11, 1.1.12

Error:There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)

danalvarez, 2021-08-17:
@jwoolaway , the solution for now is to downgrade VSCode to 1.58.2 as shown here: https://github.com/pycom/pymakr-vsc/issues/144#issuecomment-895851438

> Hello, I am havng the same error as many others with Pymakr on VScode, I am working on OSx 11.5.1 (20G80)
>
> Version: 1.59.0
> Commit: 379476f0e13988d90fab105c5c19e7abc8b1dea8
> Date: [...]

jakobrosenberg, 2021-09-03:
Hi everyone and thanks for all the feedback.

We've been working on a fix, which will be released as soon as it passes internal testing.

jakobrosenberg, 2021-09-06:
We released `v.1.1.13` which now compiles `serialport` against the Electron version of your local VSCode installation. This ensures that your plugin always has the correct binaries for your VSCode.

**PLEASE NOTE** that to make this work, `npm` has been added as a temporary dependency for the plugin to work. Since `npm` comes bundled with `node`, this shouldn't be an issue, but there could be edge cases. 

To handle these edge cases, future versions will reintroduce precompiled bindings and only use npm as a fallback for missing bindings.

---

**TL:DR the `serialport` issue should be permanently fixed. *knocks on wood***

---

Closing this issue for now as the issue is considered solved, but feel free to keep the conversation going. Thanks again for all the feedback and patience. 🙏

sltlyldfshnd, 2021-09-08:
Plugin version 1.1.13 not working on Windows 10, Vscode 1.60 - I've included the debug console output:

```
workbench.desktop.main.js:71 [Extension Host] (node:12584) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.(Use `Code --trace-deprecation ...` to show where the warning was created)
v @ workbench.desktop.main.js:71
workbench.desktop.main.js:1992 Activating extension 'pycom.pymakr' failed: Command failed: npx electron-rebuild -v 13.1.8
- Searching dependency tree
× Rebuild Failed
An unhandled error occurred inside electron-rebuildnode-gyp failed to rebuild 'c:\Users\*sanitised*\.vscode\extensions\pycom.pymakr-1.1.13\node_modules\@serialport\bindings'.For more information, rerun with the DEBUG environment variable set to "electron-rebuild".Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1Error: node-gyp failed to rebuild 'c:\Users\*sanitised*\.vscode\extensions\pycom.pymakr-1.1.13\node_modules\@serialport\bindings'.
For more information, rerun with the DEBUG environment variable set to "electron-rebuild".Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1    at NodeGyp.rebuildModule (C:\Users\*sanitised*\AppData\Roaming\npm-cache\_npx\10368\node_modules\electron-rebuild\lib\src\module-type\node-gyp.js:109:19)    at processTicksAndRejections (internal/process/task_queues.js:95:5)    at async ModuleRebuilder.rebuildNodeGypModule (C:\Users\*sanitised*\AppData\Roaming\npm-cache\_npx\10368\node_modules\electron-rebuild\lib\src\module-rebuilder.js:94:9)    at async Rebuilder.rebuildModuleAt (C:\Users\*sanitised*\AppData\Roaming\npm-cache\_npx\10368\node_modules\electron-rebuild\lib\src\rebuild.js:226:9)    at async Rebuilder.rebuild (C:\Users\*sanitised*\AppData\Roaming\npm-cache\_npx\10368\node_modules\electron-rebuild\lib\src\rebuild.js:184:17)    at async C:\Users\*sanitised*\AppData\Roaming\npm-cache\_npx\10368\node_modules\electron-rebuild\lib\src\cli.js:154:9.
$onExtensionActivationError @ workbench.desktop.main.js:1992
DevTools failed to load source map: Could not load content for https://ticino.blob.core.windows.net/sourcemaps/e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff/core/vs/workbench/workbench.desktop.main.js.map: Load canceled due to load timeout
```

Josverl, 2021-09-08:
@jakobrosenberg ,
as can be noticed above , and in a simple test , most ( 90%+) of windows installed do not have the compiler and toolchains assumed to be in place when attempting to rebuild. 
And hence the rebuild will fail for most windows users.

based on that my vote is to re-open this or another issue.
at least do clearly document the requirements to run pymaker 
just for serialport the following is needed : 
- Separate Node installation 
- the appropriate windows configuration 
- (windows-build-tools](https://www.npmjs.com/package/windows-build-tools) 
   - c++ compiler 
   - separate/supporting python installation

jakobrosenberg, 2021-09-08:
@Josverl much appreciated. Getting the prebuilt bindings automated is currently my #1 priority and I'll get back to you as soon as I have something to report.

sltlyldfshnd, 2021-09-09:
Thanks. In the meantime, it would be helpful if someone could explain how to do it manually. I do have Node & the build tools installed.

jakobrosenberg, 2021-09-09:
@sltlyldfshnd 
there's a PR that aims to fix the build deps issue: https://github.com/pycom/pymakr-vsc/pull/158 . If you clone the branch, you should be able to create the beta extension locally.

I know this is a bit of a hassle, but, for what it's worth, we're currently looking into the prospect of making these beta releases publicly available in the future.

jakobrosenberg, 2021-09-09:
Quick follow up. We've released a beta to address this issue. Please see https://github.com/pycom/pymakr-vsc/issues/157#issuecomment-915973179 for more info.

simeneide, 2022-04-04:
any updates on this? Ive followed https://github.com/pycom/pymakr-vsc/issues/130#issuecomment-797445190 for some time, but it broke today and I cant get that to work again...

jakobrosenberg, 2022-04-04:
@simeneide , We're working on a fix for this bug and will release a new version as soon as its solved. In the meantime, my best advice is to either:
- temporarily revert VSCode to the previously installed version.
- try the new [Pymakr 2 preview](https://marketplace.visualstudio.com/items?itemName=pycom.pymakr-preview).

marosokn, 2022-05-02:
@jakobrosenberg , Hello! new to the thread. Is there any update on this? I have tried the most up to date installations on node, pymakr, and vscode with no luck and I have been testing each version sequentially with no luck either. I have tried previous versions of VScode and pymakr that people have said they have had success with but with no luck. Still stuck on ">Failed to connect (Error: Port is not open) Click here to try again"

jakobrosenberg, 2022-05-03:
@marosokn it sounds like the issue lies outside of Pymakr if you are unable to connect to the device through Putty.

Often when devices won't connect it's caused by another application accessing the device. If this is not the case, I recommend contacting support to verify if the device is working.

Josverl, 2022-05-03:
@marosokn , 
could your boot.py / main.py code : 

the board does seem to start up normally `rst:0x1 [(POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-log)

- default to a different baudrate 
- or have redirected / disabled the output to a different uart ?

Can `esptool.py -p <PORT> --chip-id` access the board  ?
otherwise re-flashing the firmware will usually solve problems, but you'll loose all  info in flash