pycom/pymakr-vsc #122: Failed to connect (Error: Port is not open) message

RoAKr, 2021-02-10:
Hi, and hello, 
since some days I'm facing issues connecting my Lopy.
And after three days of searching... 

## What are the steps to reproduce this issue?
1. tried it by UART-Adapter
2. tried it by USB connection to Expansion V3.1 board
3. tried the UART-Adapter in backloop-test with another terminal. 

## What happens?
This happens: 
Connecting to COM6...
> Failed to connect (Error: Port is not open). Click here to try again.

## What were you expecting to happen?
I'm expecting to connect to the board and see the terminal output. 

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## What versions of software are you using?

**Operating system:**
MS Win 10
**VSCode version:**
1.53.1
**Pymakr version:**
1.18

elmasria, 2021-02-10:
Hi @RoAKr 

Thank you for reporting this.

We are facing an issue with new updates of VSCode (Electron 11) it is not compatible with some libraries. 

we are working to solve that issue. will update you once we found a solution

Have a nice day.

Best Regards,
Ahmad EL Masri

AntonisPlatis92, 2021-02-10:
Hello,

I am facing the same issue since some days ago, I guess after the latest update. 

I tried to downgrade to 1.1.7, but the extension is not loading. Do you know any workaround on this, until the issue is fixed?

Thanks in advance,
Antonis

elmasria, 2021-02-10:
@AntonisPlatis92 

If you want to use version 1.1.7 downgrade the VSCode to version 1.52.1. Pymakr should work properly with that setup.

Best Regards,
Ahmad EL Masri

RoAKr, 2021-02-10:
Thanks for the quick reply.
I will try a downgrade of VSCode to 1.52.1 and 1.1.7. 

Best regards,
Roman

snympi, 2021-02-11:
Have a look at what @chriswood did to fix Pico-Go ([https://github.com/cpwood/Pico-Go]) (which is built on PyMakr) to fix the issue on VSCode 1.53.0. Experienced same Port not open issue on my Pi Pico and the updates to Pico-Go. He had to update com ports but I can't explain the detail.

flokain, 2021-02-11:
i had the same issue running on ubuntu 20.04.2
downgrading code to  1.52.1 and pymakr to 1.1.7 solved the issue.

lixas, 2021-02-13:
downgrading VSCode to 1.52.1 and pymakr to 1.1.7 also worked on Win10 (19041.746)

ehtnevets, 2021-02-15:
I installed Pico-Go and it works like a charm.

elmasria, 2021-02-16:
Hi all, 

We have deployed a new version 1.1.9 which should solve the issue. 

Thank you for your feedback 

Best Regards,
Ahmad El Masri

metachris, 2021-02-16:
Still seeing the same issue with Pymakr v1.1.10

elmasria, 2021-02-16:
@metachris which os are you using?

metachris, 2021-02-16:
macOS 11.2.1

ehtnevets, 2021-02-16:
1.1.10 works for me. It's just that i had to unplug and plug the usb port after installation.

Josverl, 2021-02-17:
🎉🎉 1.1.11 work on Win-x64

Thanks @elmasria

ShettyHarapanahalli, 2021-03-29:
I am using ubuntu 20 and Pymakr v1.1.11.It fails for me and workarounds.

elmasria, 2021-03-31:
Hi @ShettyHarapanahalli , which version of VS-code are you using?

ShettyHarapanahalli, 2021-04-01:
Hi,

I had "1.54.X" earlier now I have "1.55.0" and it's working.

Best Regards,
Suman

closed 2021-05-06 by elmasria