pycom/pymakr-vsc #130: Plugin breaks when running on VS Code native ARM build on Apple Silicon

jabelone, 2021-03-12:
## What happens?
I have a MacBook Air with an Apple Silicon M1 chip in it. After updating to the latest version of VS Code (which is the first native ARM build) I can no longer use pymakr and receive an error that says "There was an error with your serialport module, Pymakr will likely not work properly. Please try to install again or report an issue on our github (see developer console for details)".

This is for a personal project, but we've used pycom boards at work before so I hope this is resolved before we get another project that uses them.

## What were you expecting to happen?
I expected the add-on to load successfully.

## Any logs, error output, etc?
```
Activating extension 'pycom.pymakr' failed: dlopen(/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.11/node_modules/@serialport/bindings/build/Release/bindings.node, 1): no suitable image found.  Did find:
	/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.11/node_modules/@serialport/bindings/build/Release/bindings.node: mach-o, but wrong architecture
	/Users/jabelone/.vscode/extensions/pycom.pymakr-1.1.11/node_modules/@serialport/bindings/build/Release/bindings.node: mach-o, but wrong architecture.
```

## Any other comments?
I believe this is because now that VS Code is running natively, the add-on is no longer running through the Rosetta 2 translation layer. I've tried to run `npm install serial port` from within the `.vscode/extensions/pycom.pymakr-1.1.11` folder through Rosetta, and tried an electron rebuild, but that doesn't seem to work. I don't know much about PyMakr or serialport internally, but I'd guess that one of them is trying to load an AMD64/x86 binding instead of an ARM64 binding. I am unsure if this is a PyMakr or a serialport issue, so feel free to tell me to file this with serialport if that's where the problem is.

## What versions of software are you using?

**Operating system:** 11.2.2 (Big Sur)

**VSCode version:** 1.54.1

**Pymakr version:** 1.1.11

jabelone, 2021-03-12:
I've just performed the following steps and confirm that PyMakr is working correctly on the same system:
1) Download the "intel" build of VS Code from the VS Code website.
2) Open and run the "intel" binary (which automatically uses the Rosetta 2 translation layer).
3) Uninstall the PyMakr binary and delete the `.vscode/extensions/pycom.pymakr-1.1.11/` folder.
4) Install the PyMakr add-on as normal via the add-on store.
5) PyMakr opens and runs fine. 

Due to the way applications work on macOS (generally being a self contained binary) I can just run the "intel" version separately whenever I need to use PyMark so it's not a huge deal, just inconvenient. Thanks in advance!

Josverl, 2021-05-06:
@jabelone , 
I think it makes sense to raise this with serialport as that is where the native bindings are sources from.
as far as I can tell they do not have any specific M1 builds

closed 2021-05-17 by elmasria