pycom/pymakr-vsc #133: Pymakr does not show in 'Code - OSS' version of VS Code

GregTheHun, 2021-04-14:
## What are the steps to reproduce this issue?
1. Open VS Code (Code - OSS version on Linux [Manjaro])
2.Try to search in Extensions tab for 'Pymakr'

## What happens?
Doesn't appear for installation

## What were you expecting to happen?
Have Pymakr appear as an installation option for the Code - OSS version of VS Code

## Any logs, error output, etc?
None, unfortunately it searches, but nothing appears

## What versions of software are you using?

**Operating system:**
Manjaro
kernel: 5.10.26-1
DISTRIB_RELEASE=21.0.1
DISTRIB_CODENAME=Ornara

**VSCode version:**
Code - OSS
Ver. 1.54.3

**Pymakr version:**
None appears

GregTheHun, 2021-04-16:
If it helps (maybe not much), but here's a screenshot:

[image]

elmasria, 2021-04-18:
@GregTheHun 

Please can you check again, I just double-checked from my side and I can see pymakr on VSC.

Thank you.

GregTheHun, 2021-04-19:
Correct, it is on VS Code one from Microsoft, but there's also an open source version you can find on many Linux distrobutions titled, "Code - OSS", that one still doesn't present it.

elmasria, 2021-05-06:
Hello @GregTheHun,

Sorry for this delay. I was checking and I saw that the Marketplace of **VSCode** is not the same as **Code - OSS**, or at least that what I have understood. 

What I did is that I create a new Release where you can have access to the `vsix` extension so you can install **Pymkr** manually. 

NB: if you have some reference on how we should add **Pymakr** to that marketplace, I will appreciate your support.

Reference: https://github.com/microsoft/vscode/wiki/Differences-between-the-repository-and-Visual-Studio-Code

[image]

Hope the above is useful. Have a nice day,

Best Regards,
Ahmad EL Masri

closed 2021-05-17 by elmasria