pycom/pymakr-vsc #135: Pymakr  v1.1.11 stop working with new version of VSC 1.56.0.

livius2, 2021-05-06:
Pymakr  v1.1.11 stop working with new version of VSC 1.56.0.

VSC Version: 1.56.0 (user setup)
Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
Date: 2021-05-04T22:09:06.405Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Windows_NT x64 10.0.19041

Pymakr version: v1.1.11

There was an error with your serialport module, Pymakr will likely not work properly. 
Please try to install again or report an issue on our github (see developer console for details)
reinstalled but did not help.

elmasria, 2021-05-06:
@livius2, Thanks for reporting this I will take a look.

tibuski, 2021-05-06:
Same issue since this morning and Visual Studio Code update 1.56.0

elmasria, 2021-05-06:
@livius2 and @tibuski We published new version 1.1.12. Let me know if you have any issues with the new version. Thank you

tibuski, 2021-05-06:
Wow !

What a service :) Issue at 8am and solved at 2pm !

Thank you very much !

livius2, 2021-05-06:
Yes, it is fixed, thx

elmasria, 2021-05-06:
Great, I am happy that it has been fixed for you Thanks for the feedback. I will close this issue.

closed 2021-05-06 by elmasria