pycom/pymakr-vsc #139: error with serialport module on Apple M1

hermankongen, 2021-05-20:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

##
[image]
 What are the steps to reproduce this issue?
1.instaalled the extension 
2.erroer appers
3.

## What happens?

## What were you expecting to happen?

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## What versions of software are you using?

**Operating system: mac os big sur 11.3.1**

**VSCode version:Version: 1.56.2 (Universal)
Commit: 054a9295330880ed74ceaedda236253b4f39a335
Date: 2021-05-12T17:44:30.902Z (1 wk ago)
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Darwin arm64 20.4.0**

**Pymakr version:v1.1.12**

hermankongen, 2021-05-20:
yolo i am going to try using atom

simeneide, 2021-05-25:
I got the same problem. did you solve it just used atom?

hermankongen, 2021-05-27:
i just switched to atom but it still has sissues but that is just elctron being electron

jakobrosenberg, 2021-09-03:
Closing in favor of https://github.com/pycom/pymakr-vsc/issues/119

closed 2021-09-03 by jakobrosenberg