pycom/pymakr-vsc #142: Pymakr doesnt work in workspace

WerdoxDev, 2021-07-26:
(Hi! 👋  Thanks for reporting an issue! Please make sure you click the link above to view the issue guidelines, then fill out the blanks below.)

## What are the steps to reproduce this issue?
1. just create a workspace and add few folders
2. then try to upload

## What happens?
pymakr tries to find the sync_folder only in the first folder of the workspace
well i have the project in the second folder but it just doesn't look for it (the first one is a web app)

## What were you expecting to happen?
I would expect it to only try find the sync_folder inside the folder which has pymakr.json in it or some way of
choosing which folder of workspace
or just search the whole workspace which might not work when having multiple pymakr projects in a workspace

## Any logs, error output, etc?
not really it just doesn't look for other folders in the workspace and when it doesn't find any in the first folder it gives this error
which is not the problem

"Upload failed.: Unable to find folder 'srcp' in your project. Please add the correct folder in your settings"

## Any other comments?
im using Micropython and a esp8266 board

## What versions of software are you using?
not using the pycom

**Operating system:**
Ubuntu 21.04

**VSCode version:**
latest

**Pymakr version:**
latest

WerdoxDev, 2021-07-26:
for now i just put the pymakr project in the first folder but it would be nice to not have this problem since i like to organize the order of the folders from most important to least

jakobrosenberg, 2022-04-08:
This should be solved in Pymakr 2.

closed 2022-04-08 by jakobrosenberg